Serial-Studio/Serial-Studio #115: Compilation fails on updated Arch system

mcgdb, 2022-07-26:
## Describe the bug

Compilation fails because it can't find `QStringConverter`

### Screenshots
```
compiling libs/qmqtt/src/mqtt/qmqtt_client_p.cpp
In file included from libs/qtcsv/sources/reader.cpp:1:
libs/qtcsv/include/qtcsv/reader.h:9:10: fatal error: QStringConverter: No such file or directory
    9 | #include <QStringConverter>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

## Upload your JSON map file*
N/A

## Upload minimal Arduino program to replicate your issue

N/A

Redika-ak, 2022-11-22:
So Do I

pwolny, 2023-01-01:
I had the same issue when trying to compile Serial-Studio on debian bullseye.
After I started using a  qt4_qt5 branch of qtcsv  
( git clone --branch qt4_qt5   "https://github.com/iamantony/qtcsv.git" )
this specific error went away (still had other errors when compiling).

alex-spataru, 2023-01-22:
Hi! Sorry for the (very) late response, I had a hell of a year (2022), and between finishing college, working my thesis, and attending two jobs, I had very little time left to check my e-mail or respond to bugs/issues.

Sometime during last year, Serial Studio was updated to be compiled with Qt 6. However, I forgot to update the README.md file and work on updating the wiki/documentation. 

I´ll update the documentation during the week. I´ll notify you as soon as the new compilation instructions are available.

closed 2024-11-17 by alex-spataru