Serial-Studio/Serial-Studio #118: 功能增强请求

zhigangbox, 2022-08-11:
serial studio在可视化方面很简单便捷；microlab显示组件和控制组件很丰富。那么serial studio能否借鉴microlab的特点
1 让显示组件更丰富：比如进度条，刻度仪，LED,数码管。
2能否增加控制组件以便于向下下发指令：按钮，开关，滑动条，旋钮
3 是否有个论坛和聊天群一块交流

alex-spataru, 2022-09-06:
Hi! Thanks a lot for your interest in this project! I have never used Microlab, could you please send me a link to try it? I think that most of the widgets that you mentioned can be implemented in future releases, I'll keep you updated through this thread.

For the moment, the only available forums/chat groups are on GitHub ([issues](https://github.com/Serial-Studio/Serial-Studio/issues) and [discussions](https://github.com/Serial-Studio/Serial-Studio/discussions)). I tried to use a [Gitter](http://gitter.im/serial-studio) channel for a while, but nobody participated there 😆.

If you have any suggestions/ideas for the communication channel please let me know!

---

你好！ 非常感谢您对这个项目的兴趣！ 我没用过Microlab，能发个链接给我试试吗？ 我认为您提到的大多数小部件都可以在未来的版本中实现，我会通过这个线程让您保持更新。

目前，唯一可用的论坛/聊天组在 GitHub ([issues](https://github.com/Serial-Studio/Serial-Studio/issues) 和 [discussions](https://github.com/Serial-Studio/Serial-Studio/discussions)). 我尝试使用 [Gitter](http://gitter.im/serial-studio) 频道一段时间，但没有人参与其中😆。

如果您对沟通渠道有任何建议/想法，请告诉我！

evalor, 2023-01-15:
I'm also very interested in this. A useful example is [Fuax](https://github.com/frangoteam/FUXA)

It's a web-based Process Visualization (SCADA/HMI/Dashboard) software. I have used it in many Internet of Things projects. There is much to learn from.

closed 2024-11-17 by alex-spataru