Serial-Studio/Serial-Studio #120: Manually Set Y-Axis Limits on Graphs

SammiMD, 2022-08-31:
Is there a way to manually set the Y-Axis limits on graphs/plots?

I'm measuring pressure, and because my readings all sit around the 1000Pa mark, my plots are coming out as straight lines. I'd like to be able to see the minute fluctuations in the sensors' readings in a plot.

[image]

TIA

alex-spataru, 2022-09-06:
Hi, you can modify the Y-axis limits by setting the minimum and maximum values for each dataset in the JSON editor (see screenshot below):

[image]

Please let me know if this helped you!

phatpaul, 2024-09-16:
Any plans to make this adjustable (i.e. with the mouse-wheel) at run-time?

alex-spataru, 2024-11-17:
> Any plans to make this adjustable (i.e. with the mouse-wheel) at run-time?

Maybe in the future. I think it would be useful to add a window that allows "scaling" signals in a similar way to how you would do it in an oscilloscope.

closed 2024-11-25 by alex-spataru