Serial-Studio/Serial-Studio #121: UI update.

goutham03062001, 2022-09-04:
I just saw the UI of  serial studio platform. And I thought it would be very attractive and beautiful if we can modify it's User Interface. 

And I'm currently working on it. I would like to take suggestions from other developers. 

What's ur thoughts on this?

alex-spataru, 2022-09-06:
Hi! Thanks a lot for your interest in this project! The user interface can be customized by creating and modifying [themes](https://github.com/Serial-Studio/Serial-Studio/tree/master/assets/themes). However, for the moment, themes can only define the colors of the application. But I think that this functionality can be extended with the appropriate development effort.

Please let me know if I can help you with anything regarding your work, you are very welcome to open a pull request!

goutham03062001, 2022-09-06:
Thanks for your concern alex-spataru. 

I didn't get you the "this functionality" means. Can you please let me know if you don't mind.

closed 2024-11-17 by alex-spataru