Serial-Studio/Serial-Studio #124: Serial Port Connections in Windows 11

IrisEmbedded-DM, 2022-09-05:
### Discussed in https://github.com/Serial-Studio/Serial-Studio/discussions/123

<div type='discussions-op-text'>

<sup>Originally posted by **IrisEmbedded-DM** September  5, 2022</sup>
I'm currently testing out Serial Studio for visualising data from my embedded sensors, running on Windows 11. My COM port displays in the list, but when I connect, the data only displays maybe once every 20 times I launch the software. I've pared the data back to be as simple as possible (just CSV with a newline) and yet this still rarely displays in the console. I have also tried setting up frames and the JSON, but this always leads to no data.

Checking the COM port with other software shows that the data is there! 
[image]
[image]
</div>

alex-spataru, 2022-09-06:
Hi! Thanks a lot for the report, I will do some tests on Windows 11 and get back to you. For the moment, could you please try the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) and let me know if the issue persists?

IrisEmbedded-DM, 2022-09-06:
The issue is still there with the latest build unfortunately! The serial data should still display in the console regardless of any JSON parsing right?

recursivenomad, 2024-02-10:
I believe this is a problem for Windows builds in general, as it is reproducible on multiple Windows 10 machines as well.

See here:
- #124
- #135 
- #98

recursivenomad, 2024-06-14:
This problem appears to be isolated to specific microcontrollers, and only present on Windows (as mentioned in [#135](https://github.com/Serial-Studio/Serial-Studio/issues/135#issuecomment-1937016698)).  The issue appears to be that some sort of initial handshake is missing for certain microcontrollers, because when you initialize the serial connection through another serial monitor (such as the Arduino IDE, PuTTY, etc), and close that serial monitor, Serial Studio then can successfully connect and reads back the data as desired.  The problem presents itself again each time the device reconnects.

I can confirm Serial Studio 2.0.0 still does not work as intended when attempting to display serial output from the Raspberry Pi Pico or Arduino Pro Micro on Windows 10.  [Previous reports](https://github.com/Serial-Studio/Serial-Studio/issues/98#issue-1139835261) have included the Arduino Nano Every as having the same problem as well, and this thread reports it existing in Windows 11.

I *have* been able to get a XIAO-ESP32-S3 **working** successfully with Serial Studio 2.0.0 on Windows, so the problem does not appear to be my machine, but rather the combination of the Windows OS and certain popular microcontrollers.

alex-spataru, 2024-08-05:
Thanks for @recursivenomad for adding a possible fix for this issue. The latest build adds a checkbox that allows users to send the DTR (Data Terminal Ready) signal upon connecting to a serial port device:

[image]

Please let me know if the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build fixes this issue!

JustTheCoolest, 2024-09-25:
I think I have the same issue.

On the continuous build, I'm now able to connect to my Arduino Uno (finally). But even with the same json and data as the communication protocol example, I'm not able to get the dashboard open. 

Am I doing anything wrong?

(Also, please consider making a video tutorial of the basic usage. I'm very clueless on whether I'm getting the current responses from the app 😅 )

[image]

JustTheCoolest, 2024-09-25:
Same in Linux (WSL2 Ubuntu):

[image]
[image]

alex-spataru, 2024-09-26:
**@JustTheCoolest** You're using an outdated project file format—my apologies for the confusion, as I haven’t had the chance to update the wiki yet. To create a JSON project, there’s no need to manually write any code. Instead, I recommend using the built-in Project Editor, which you can access via the toolbar buttons.

You might also want to try the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), where I’ve made significant improvements to the user interface to make the application more intuitive and user-friendly.

alex-spataru, 2024-09-26:
By the way, please take a look at [this comment](https://github.com/Serial-Studio/Serial-Studio/issues/205#issuecomment-2375490542) for an updated project file and code example. In this case, the Arduino code is slightly more complicated, as it demonstrates how to use user-defined action buttons to trigger specific behaviors on the microcontroller.

JustTheCoolest, 2024-09-26:
Working now!
For the sake of documenting on the GitHub repo, here is a video of it working on Windows:

https://github.com/user-attachments/assets/1c243633-30e8-4c9f-abef-7dcfcc59be87

alex-spataru, 2024-09-26:
Great to hear that, and thanks for the video! On my end, I’m planning to re-write the entire wiki to reflect the current release I’m working on. 

By the way, have you had a chance to try out the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)? Your feedback would be really valuable in helping me catch any remaining bugs or missing features before I dive into updating the wiki and finalizing the release.