Serial-Studio/Serial-Studio #125: App crashing when receiving high-speed data stream on serial port

pedroavex, 2022-09-05:
Hola Alex! Que tal? This is Pedro from Portugal and i am trying to use your nice piece of SW to display some live data from a CANBUS collector that i have integrated in my ESP32 project, that also collect simultaneously other GPIO connected to other car sensors.
The problem is that the string rate coming from ESP32 to Windows 10 64bit COM port is quite high, something between 1,000 to 3,000 strings/seconds. The strings are in readable text and they look like this:
3714254,3,8,aa,aa,aa,aa,aa,aa,aa,aa
3714938,4,8,bb,bb,bb,bb,bb,bb,bb,bb
3715639,3,8,aa,aa,aa,aa,aa,aa,aa,aa
3716312,4,8,bb,bb,bb,bb,bb,bb,bb,bb
3717286,1,8,0,82,0,82,0,82,0,82
3717426,2,5,1,1,1,0,0
3717609,3,8,aa,aa,aa,aa,aa,aa,aa,aa
........
I am not using any Jason file to parse this data yet because i'm in learning process, but the fact is that when i open the COM port, which i run at 1Mbps baudrate from ESP32, the Serial Studio works for some seconds than starts to freeze, freeze up to the point it becomes not responsible anymore.
I have seen another friend talking about maximum speed and crashing the same way, i don't know is it is the same case, but...
P.S. In a simple terminal software or even in Arduino IDE serial monitor i can see the data flowing without problem at this baud, so the bottleneck is not the serial port or baud, for sure.
Thanks!

alex-spataru, 2022-09-06:
Hi, I think this has something to do with the UI rendering thread. I will do some tests on my own computer and get back to you in a few days. Thanks a lot for the bug report!

pedroavex, 2022-09-06:
@alex-spataru I had a feeling that it could be related to I/O HAL or some buffering issue, because what i have noticed is that the crash doesn't happen at once, it starts fine, then after some seconds shows some freezes, these freezes starts taking longer and longer up to the point the program becomes not responsive at all. It is a progressive situation.
I have noticed also that overall the program shows some "blinks" even before i open the serial stream. I just start the program and sometimes the the window disappears for a couple of millis and come back, almost like a blink of an eye, but happens.
Maybe it has to do witih my machine and graphics card, i don't know. Anyway, this is my Config:
Windows 10 Home 64bits 21H2, AMD Ryzen 9 3900X 12-Core 3.8GHz, 32GB RAM, G-Force RTX2060
One more small detail: the splash screen when i open the program is still showing version 1.1.16, i thought i had installed the wrong version but no, it was just the splash screen
But anyway, you are the father of this beautiful child, you are the best to find out what's going on. I have made a small video for it, please check. Thanks!!

alex-spataru, 2024-11-17:
Hi @pedroavex, you might want to check out the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous). Since this issue was opened, there have been major improvements, and I’ve been working specifically to address the high-frequency input problem that was overloading the IO module.

The latest continuous build of Serial Studio can now handle data intervals of less than 1 ms between frames, as demonstrated in the [UDP Function Generator example](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/UDP%20Function%20Generator).

Since version 3.0, UI rendering is capped at 24 Hz to ensure smooth performance even on older hardware and virtual machines (tested on pre-2012 systems). The plots and CSV files generated still reflect all received data, so no information is lost.

One caveat of the threaded/queued approach is that RAM usage may increase gradually if the input data rate exceeds what Serial Studio can process/consume or what the operating system's scheduler permits, not too much I can do about that unfortunately. 

I aim for Serial Studio to eventually handle data with a sleep cycle as short as 0 ns from local processes gracefully, though “remote” devices like serial ports, network hosts, or BLE devices will always introduce some inherent delay.  I’ll keep refining this, but for now, it’s a solid step forward. Thanks for your patience!

closed 2024-11-17 by alex-spataru