Serial-Studio/Serial-Studio #127: Can’t load the dashboard and can’t even edit my json

WOD-MN, 2022-09-15:
## Describe the bug
[image]

[image]

even i have used examples , i couldn’t load the dashboard!!!
Please help

yukelab, 2022-09-20:
maybe, you an see issues 101 https://github.com/Serial-Studio/Serial-Studio/issues/101

closed 2024-11-17 by alex-spataru

WOD-MN, 2024-12-22:
ok

alex-spataru, 2024-12-22:
@WOD-MN Sorry for closing the issue without responding. The project has changed a lot since this issue was opened and it did not make sense to add an outdated answer in the discussion. Please check the renewed [wiki](https://github.com/Serial-Studio/Serial-Studio/wiki) for guidance on how to setup a project.