Serial-Studio/Serial-Studio #129: Dark mode for Windows

IsaacBlancICL, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
The screenshots on the Serial Studio website show the program operating in dark mode (on Mac OS), which I think looks much nicer than the non dark mode Windows version.

**Describe the solution you'd like**
Dark mode for Windows too, please!

alex-spataru, 2022-10-28:
Hi!

You can change Serial Studio's theme (regardless of the operating system that you are using) by navigating to the "Settings" tab and selecting your desired theme. 

[image]

A good idea for the future would be to replicate Window's dark color scheme for a better integration with the rest of the system...

Please let me know if this fixes your issue!

IsaacBlancICL, 2022-10-28:
Thank you for this. I completely missed that!

closed 2023-01-22 by alex-spataru