Serial-Studio/Serial-Studio #130: Qt version to compile the project

JesseGuoX, 2022-10-30:
## Describe the bug

Hi there.
   Is this project only compiled with Qt6 ? When I compile with Qt5.15 there are a lot of errors about "package import requires a version number", and I think it is a Qt6 feature  which can ignore the module version.

### Screenshots

[image]

gremsyQuanLe, 2022-10-31:
[image]
I'm Qt 5.15.2 error build

alex-spataru, 2024-10-31:
Serial Studio is now dependent on Qt 6.8 LTS version, compiling with Qt 5 is not supported (you might be able to work with it, given you change enough things on the QML side). Closing this issue for now.

closed 2024-10-31 by alex-spataru