Serial-Studio/Serial-Studio #134: Support HEX data frame

Yurisu, 2022-12-13:
1. Support HEX frame start sequence and frame end sequence
2. Customization end line break.(use HEX)
4. Separated by sequence in HEX.
5. Support data show in single byte , double byte , Quad bytes....
6. Customization new queue timer.(use low baud rate , like 4800 / 2400)

evalor, 2023-01-15:
This is great. I have a frame format, for example: 0xFA7F000085

1. Frame always starts with 0xFA
2. Fixed length, 5 bytes
3. The last byte is XOR verification except for itself.

It would be great if serial studio could work for me.

alex-spataru, 2023-01-22:
Hi! You might want to try the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) in order to parse hex data.

This build allows users to input custom frame-parsing code for project-specific needs. You can modify the frame parser code by clicking on the "Customize frame parser" button in the Project Editor:

[image]

Afterward, you will be presented with a JavaScript code editor:

[image]

**NOTE:** start/end sequences must be strings constructed with ASCII characters. I will update the code to support inputting HEX data/numbers during the course of the week.

Thanks for the idea!

mfvmx, 2023-02-09:
We have the same thing (hex) but with a variable length, where the second value is length to indicate how many tempurature values will be incoming.

alex-spataru, 2024-10-31:
> Hello, this feature already enable in latest build? if yes, how can I use it?

Hello! With the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0), it’s now possible to work with Hexadecimal data in Serial Studio. You can find an example [here](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/HexadecimalADC). I’ll close this issue for now, but feel free to let me know if it should be re-opened.

closed 2024-10-31 by alex-spataru