Serial-Studio/Serial-Studio #135: Pi Pico USB serial port reconnect issue

bseaby, 2022-12-16:
I have Pi Pico outputting JSON data via its USB serial port. If I open putty and conenct, I get the serial stream. If I then open the port in Serial Studio, I get JSON data in the console, and in the dashboard, it works great.

However, if I then unplug and reconnect the USB device, and reconnect in Serial Studio, I get no data in the console. I have to disconnect Serial Studio, open putty, connect and then close putty, then press connect in Serial Studio in order to fix the issue.
I've tried this on 2 different machines (Both Win 10 Version	10.0.19042 Build 19042), and have tried both the latest version of serial studio and version 1.1.4. Not tried in linux yet.

recursivenomad, 2024-01-21:
I'd just like to chime in that I am also experiencing this behaviour on Windows.

I am attempting to view the Raspberry Pi Pico raw serial stream (following [this DigiKey tutorial](https://www.digikey.com/en/maker/projects/raspberry-pi-pico-and-rp2040-cc-part-1-blink-and-vs-code/7102fb8bca95452e9df6150f39ae8422)), but when I try to use Serial Studio instead of PuTTY, it is sometimes able to "Connect", but never streams any raw serial data.  Only after successfully opening the COM port in PuTTY and seeing the serial stream, closing PuTTY, and then selecting the COM port in Serial Studio am I able to connect and view the raw serial stream.

PeterSHollander, 2024-02-10:
This may be Windows-specific, I am not seeing this behaviour on Debian.

recursivenomad, 2024-02-10:
This may be Windows-specific, I am not seeing this behaviour on Debian.

closed 2024-08-05 by alex-spataru