Serial-Studio/Serial-Studio #136: Data appearing on console only.Dashboard greyed out

naubeeluck, 2022-12-23:
Hello,
I am receiving data via serial port but can only see the data on the console. The dashboard does not appear with the real time plots of the data and is greyed out? Please advise.

anthoine001, 2023-03-09:
Hello. After making several tests it seems the problem appears on windows10 : I run the program on Mac and there is no issue. Could you tell us on what platform did you get the problem ?

sunw1206, 2023-03-14:
Hello! I have the same problem. The dashboard button always be grey. My computer's platform is windows 10.

ElfyShort, 2023-11-02:
Software can not be used unless this bug is fixed.
Unfortunate.

alex-spataru, 2023-11-02:
Hi, can you please share with me both your JSON project(s) and a saved snippet from the console output? As @anthoine001 mentions, this issue might be specific to Windows (I generally use macOS or Linux). Also, could you please tell me which version of Serial Studio are you using? Does the issue persist on the [continous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) builds?

ElfyShort, 2023-11-02:
Yes, I tried all versions, none worked for win 10. I tried several machines with the same outcome.
The button is just not clickable. It is not related to project or unit connected. The freshly installed app when opened has greyed out dashboard.
When I connected arduino which spills out data, space or coma separated, with \n in the end, console reads the data in the expected format.
Could this be a UI bug that grey-out the button in some cases? Maybe making it active at all time will solve it?

alex-spataru, 2023-11-03:
Quick question: did you create and load a project file? Serial Studio needs to use this file to know how to parse the incoming data. You can easily create new projects by clicking on the "Project Editor" button:

[image]

In the project editor, be sure to define the following properties:
- **Project name:** This parameter lets Serial Studio create a folder in the CSVs directories with the parsed data.
- **Frame start sequence:** Used by Serial Studio to know when to start reading a frame (you can set it to any character or string you want).
- **Frame end sequence:** Very similar to the start sequence, but used to know when to stop parsing a frame (e.g. you can set it to `\n` for your project).
- **Separator character:** Used to separate a dataset from another, it defaults to `,`, but you can set it to anything you want.

After that, you can click on the "Add Group" button to start defining the title (and what to do) with each dataset. You can create multiple groups to implement widgets such as accelerometers, datasets, multiple data plots, etc. You can duplicate the dataset indexes to your liking (I recommend you to keep the project short & simple so that the render loop is faster than the data frequency).

[image]

**Important:**

The frame start/end sequences are mandatory, since some users tend to send "debug" data or messages, which Serial Studio should ignore when parsing data for the dashboard. I generally tend to use the following sequences:
- *Frame start sequence:* `$`
- *Frame end sequence:* `;\n` 

An example frame sent over the serial port (or BLE, TCP/UDP, MQTT, etc) could be:

`$Reading_1, Reading_2, ..., Reading_n;`

The default sequences `/*` and `*/` are meant to replicate a multiple line comment in C/C++, however, I noticed that most people get confused with this format. I should also find the time to write good documentation in the wiki and/or create a YouTube series to ease the learning curve of Serial Studio.

Please let me know if this helped you solve your issue, or if it still persists after creating a project!

alex-spataru, 2023-11-03:
By the way, if your project file is complex, or you are dealing with binary data, you can choose to implement your own frame parsing code by clicking on the "Customize frame parser" button. Doing so will open the following window:

[image]

You can implement your own JavaScript code to let Serial Studio know how to obtain an *array* with the values in your frame, this feature is only available in the [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) for the moment.

ElfyShort, 2023-11-03:
Ok, this was helpful.
I managed to set it up and the dashboard now works.

Have 2 quick questions in return.
1. The first point I output is the time in ms. Is it possible to apply it as X on the Multiple data plot?
2. I have many datasets on the same graph. It would be great if when I hover my mouse at a certain area in the graph to see the ruler and the value that was at that point for each dataset. Serialplot is doing this, for example:
[image]

In serial studio I cant tell from the graph what data I receive, as the new data is on the right, and the scale is on the very left.

[image]

Is this is something I can turn On, or will this require extra implementation?

Thank you.

alex-spataru, 2023-11-03:
Good morning,

Glad to hear that the issue was related to the project configuration rather than the dashboard itself, I need to take the time to write proper documentation & tutorials to help new users.

> The first point I output is the time in ms. Is it possible to apply it as X on the Multiple data plot?

Currently, Serial Studio only plots X-axis data based on the sample number, not the time. I am very aware of this limitation, and plan to incorporate support for using time as the X-axis in a future release. 

Additionally, I'm considering modifying the JSON project format to allow users to specify which values should be plotted on the Y-axis and which on the X-axis. This feature could potentially help in a wider range of usage scenarios, such as joystick or sensor testing. 

I just need to find some time to re-structure Serial Studio from the ground up to be more flexible, create proper documentation and make it easier for others to contribute to the codebase (e.g. using Python instead of C++). I originally wrote this software for a very limited usage scenario when I was a student, and the project grew too fast for me to maintain it properly.

> I have many datasets on the same graph. It would be great if when I hover my mouse at a certain area in the graph to see the ruler and the value that was at that point for each dataset. Serialplot is doing this, for example:

Thank you for the suggestion! I often use a similar feature when working with matplotlib graphs. I will incorporate this feature in a future release and keep you informed once it's available.

ElfyShort, 2023-11-03:
Hi Alex,
Thank you for the feedback.
For me personally, only the second feature (with a cursor-ruler and the values) is required to start using the software in my project as I really need to see precise values of the parameters that are changing in real time.
I would really appreciate if you can let me know once it is available.
Currently I use the SerialPlot, but that guy started to crash on win10 lately (I blame windows update) and Serial Studio is basically the only alternative that I found that will work in my use case.

alex-spataru, 2023-11-07:
Hi Elfy,

Thanks for the feedback! I will post a message on this thread as soon as I implement it. Regarding the Windows 10 issue, I believe it is a common issue when dealing with serial ports. Many other users have also reported that they had problems with Windows 10 (I believe its a driver-related issue, I think it would be non-sense from Microsoft to change their serial port APIs without notifying anyone).

Thanks,
Alex

thzll, 2024-09-08:
Hello,
I am receiving data via serial port but can only see the data on the console. The dashboard does not appear with the real time plots of the data and is greyed out? Please advise. I create and load a project file.
[image]

thzll, 2024-09-08:
[image]

alex-spataru, 2024-10-31:
@thzll, @ElfyShort, @sunw1206, @anthoine001 & @naubeeluck -  Please check out the [wiki](https://github.com/Serial-Studio/Serial-Studio/wiki), sorry it got me so long to update the documentation! In many cases, you can use Quick Plot mode to avoid creating a full project—just be sure you're using the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0). Thanks!

closed 2024-10-31 by alex-spataru