Serial-Studio/Serial-Studio #137: Dashboard not showing data

cdthomp1, 2022-12-27:
## Describe the bug

I am having some trouble getting Serial Studio to display my data from my Teensy 4.0 board. I think I set up the `Serial.print` lines correctly in the code, as I can see that data flowing into the console of Serial Studio, but I don't see the data reflecting in the graphs. I attached screenshots of the empty graphs and the console showing the data. I am not sure if my JSON file is off or if it is something else entirely. 

Board: Teensy 4.0
Sensor: BNO055

### Screenshots

[image]

[image]
## JSON map file

```{
    "frameEnd": "",
    "frameStart": "",
    "groups": [{
        "datasets": [{
                "alarm": 0,
                "fft": false,
                "fftSamples": 1024,
                "graph": true,
                "led": false,
                "log": false,
                "max": 0,
                "min": 0,
                "title": "X - Acceleration",
                "units": "Meters ",
                "value": "%1",
                "widget": ""
            },
            {
                "alarm": 0,
                "fft": false,
                "fftSamples": 1024,
                "graph": true,
                "led": false,
                "log": false,
                "max": 0,
                "min": 0,
                "title": "Y - Acceleration",
                "units": "Meters ",
                "value": "%2",
                "widget": ""
            },
            {
                "alarm": 0,
                "fft": false,
                "fftSamples": 1024,
                "graph": true,
                "led": false,
                "log": false,
                "max": 0,
                "min": 0,
                "title": "Z - Acceleration",
                "units": "Meters ",
                "value": "%3",
                "widget": ""
            }
        ],
        "title": "Acceleration",
        "widget": ""
    }],
    "separator": "",
    "title": "Explorer 1v2"
}
```

## Upload minimal Arduino program to replicate your issue

My `main.cpp` file:

```cpp
#include <Arduino.h>
#include "BNO055.h"

BNO055 bno;

void setup()
{
  Serial.begin(9600);

  bno.InitializeSensor();
}

void loop()
{
  Acceleration acc = bno.GetAcceleration();

  Serial.print("/*");
  Serial.print(acc.x);
  Serial.print(",");
  Serial.print(acc.y);
  Serial.print(",");
  Serial.print(acc.z);
  Serial.print("/*");

  delay(1000);
}
```

alex-spataru, 2022-12-28:
Hi @cdthomp1!

I think that the cause of your issue is that you are using the same start and end frame sequence in your MCU code, please try re-programming your MCU with the following code:

```cpp
#include <Arduino.h>
#include "BNO055.h"

BNO055 bno;

void setup()
{
  Serial.begin(9600);

  bno.InitializeSensor();
}

void loop()
{
  Acceleration acc = bno.GetAcceleration();

  Serial.print("/*");
  Serial.print(acc.x);
  Serial.print(",");
  Serial.print(acc.y);
  Serial.print(",");
  Serial.print(acc.z);
  Serial.print("*/"); // Changed "/*" to "*/"

  delay(1000);
}
``` 

Frame start & end sequences must be different from each other to let Serial Studio know how to separate incoming data. Please let me know if this helps you!

Happy holidays,
~Alex Spataru

cdthomp1, 2022-12-28:
Thanks for the quick reply, that made it work flawlessly. Thank you!

closed 2022-12-28 by cdthomp1