Serial-Studio/Serial-Studio #138: Simple math in widgets

SergeTauger, 2023-01-10:
It's common that MCU uses internal units to make calculations more simple and fast. I.e. it is common for RTD to output abstract uint32 value that can be converted to resistance/temperature with shifts and multiplication by a constant value. However, floating point math is heavy and usually avoided. It will be perfect if there will be a possibility to multiply/divide data by a constant value prior to visualization.

alex-spataru, 2023-01-22:
Hi! Sorry for the late response (check my comment on issue #115). 

With the latest stable release, you can use some limited JS expressions to implement math operations directly in the JSON file. You can check this [issue](https://github.com/Serial-Studio/Serial-Studio/issues/29) for more information (just note that the JSON format has been extended to use more descriptive keys (e.g. "value" instead of "v").

Please note that with the next release, this feature will be disabled, since JSON project files will be able to use user-specified code for frame parsing (in which you can any math operations and/or custom data processing), I´ll notify you when I draft the new release.

For the moment, please let me know if the example on issue #29 helps you!

SergeTauger, 2023-01-22:
Thank you, just the thing I wanted. When you finish with the update, please, update wiki, too.

closed 2024-10-31 by alex-spataru