Serial-Studio/Serial-Studio #140: Support Modbus RTU and Modbus TCP

tonyho, 2023-03-12:
**Is your feature request related to a problem? Please describe.**
Modbus TCP and RTU are widely used in fieldbus and industry. 

**Describe the solution you'd like**
Qt has modbus module, project can use it to help to support it.

alex-spataru, 2024-11-26:
Hi, thank you for your suggestion, and I’m sorry for the very late response.

I’m seriously considering adding support for Modbus (and also CANbus), as I know how valuable these features are for industrial applications. However, implementing them requires access to specialized hardware for testing, which is a significant investment.

Please check this [discussion](https://github.com/Serial-Studio/Serial-Studio/discussions/234) for more details. Thank you for your patience and for supporting Serial Studio!

victorhooi, 2025-01-16:
I'm interested in this as well - however, that discussion link goes to a HTTP 404 for me?

Is there a correct link?

alex-spataru, 2025-01-16:
@victorhooi 

Hi! The discussion was a post where I invited community members to contribute to the project to support its development and ensure long-term sustainability...Unfortunately, there wasn’t any response from the community after several months.

Due to some dependencies, I had to change the license from MIT to GPL3 (#248). I’m now working on a commercial version of Serial Studio for enterprise, business, and professional users who may not be able to comply with GPL3.

I’ll keep you updated on both this issue (Modbus support) and the availability of the commercial versions.

I believe the commercial versions will not only help fund the project but also make it easier for companies to adopt Serial Studio and enable better technical support for all users.

Thanks,
Alex

victorhooi, 2025-01-16:
Of course - I understand! And I appreciate al the work you're doing in this program.

Hmm - with the Modbus stuff - I know it's a token -  but I'm happy to financially contribute towards some basic gear (I assume sensors, USB-to-RS485 adapters etc.) if that would help with your efforts? LMK.