Serial-Studio/Serial-Studio #148: serial studio crashes when it starts to parse the data

ricardo192, 2023-06-15:
readout works just fine when I turn on parsing the whole thing crashes I've tried a lot of different things and nothing really resolves this. i'm using macOS. 

	Serial.print(":");
	Serial.print(getXl_B().X); Serial.print(",");
	Serial.print(getXl_B().Y); Serial.print(",");
	Serial.print(getXl_B().Z); Serial.print(",");
        // and so on 
	Serial.print(";");

 code snipet runs at like 10 hertz I normaly would run it at 100 but 10 just to see if its a bandwidth issue 

[etherstuff copy.txt](https://github.com/Serial-Studio/Serial-Studio/files/11758872/etherstuff.copy.txt)
here's the json just change the file extension. 

I can send crash report if you want me to send it to you, I have a feeling it has something to do with the visualization of the data.

ricardo192, 2023-06-15:
weird, I made another json with the same stuff and it seems to work, it's a different number of groups though the data is arranged differently, but begs the question why did the first one not work. Anyway I got this to work but there still might be a bug in the software so I'll just leave this here 😆 .

alex-spataru, 2024-10-31:
Hi! This issue stemmed from how Serial Studio handled Qwt widget rendering on the dashboard, which led to significant performance issues, particularly on older hardware. Please try the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0) and let me know if the issue persists. In this version, I’ve made substantial improvements to the rendering process—including GPU utilization for widget and terminal rendering—to significantly reduce CPU load, prevent event thread overload, and enhance overall stability.

I’ll close this issue for now, but please feel free to re-open it if you still encounter the crash or any related issues in the latest version. Thanks, and I hope this release resolves it for you!

closed 2024-10-31 by alex-spataru