Serial-Studio/Serial-Studio #149: Serial studio window always opens half off the screen & never maximises properly. Playback box always opens off screen

quix2k23, 2023-06-16:
## Describe the bug

Every time serial studio 2.0.0 is started the main window is half visible in the bottom right of my main monitor. I need to drag it fully onto the desktop. When I try to maximize the window it reverts to being halfway off the desktop.

The playback box always opens with about 5% of it's window visible on the desktop. 

### Screenshots

[image]
[image]

## Upload your JSON map file*

[easun project.zip](https://github.com/Serial-Studio/Serial-Studio/files/11769323/easun.project.zip)

## Upload minimal Arduino program to replicate your issue
problem is not with parsing code it is with the physical layout of the UI and it's failure to maximize or display properly

dsm, 2023-09-26:
Issue is related to qt 6.5, tested qt 6.2 LTS there is no problem. This will be fixed the downgrade to qt 6.2.x

alex-spataru, 2023-09-26:
I just replicated this issue too. I will try to implement a fix for Qt 6.5 whilst maintaining compatibility with Qt 6.2. I'll get back to you as soon as I have some progress.

alex-spataru, 2024-11-07:
Quick question, is it possible for you to check if this issue still occur with the latest [release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.4)? 

I've changed a lot of things, including most of the window management code & Qt versions...at least in my test scenario I could not replicate this issue again.

dsm, 2024-11-07:
issue is fixed but window position never restored if maximized and restore down. I have two display and it going right to the center of the screens. first it goes to the position on the relevant screen and then it goes to the center of the screens.

alex-spataru, 2024-11-08:
Hi @dsm!

The latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) includes changes aimed at resolving this issue. I’ve tested it on macOS (single-screen setup), but since I don’t have a multi-monitor configuration to confirm, it would be greatly appreciated if you could try this build and let me know if the fix works on your setup too.

Thank you in advance for your help and feedback!

dsm, 2024-11-08:
Hi, @alex-spataru thanks issue solved.

quix2k23, 2024-11-08:
I installed the latest version and QT 6.8. Now serial studio refuses to start. 

Windows 11.  It's such a pity as the program is so useful. Any ideas? Is there a log in a local somewhere of what is happening after I try to start the program?

alex-spataru, 2024-11-08:
> @dsm Hi, @alex-spataru thanks issue solved.

Glad to hear that!

> @quix2k23 I installed the latest version and QT 6.8.	

If you’re not compiling Serial Studio yourself, there’s no need to install Qt separately; all necessary dependencies are included with the installer. However, you might need to install the [Microsoft Visual C++ Redistributable](https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170#latest-microsoft-visual-c-redistributable-version), which isn’t bundled with the installer. You can download it here: https://aka.ms/vs/17/release/vc_redist.x64.exe.

> The OS is Windows 11. It’s such a pity as the program is so useful. Any ideas? Is there a log in a local folder somewhere of what is happening after I try to start the program?

It’s strange that you’re experiencing this issue, as I’ve tested Serial Studio on both Windows 10 and Windows 11 (including on some older laptops over 12 years old). Could you please try the following builds and let me know if either of them resolves the issue?

- The latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous).
- The latest [stable release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.4).

Both versions display as version 3.0.4 because I update the continuous release versioning only when a new stable release is crafted. This way, continuous testers are notified about new stable releases.

In the current [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), I enabled FreeType font rendering to address issues on “legacy” displays using a 100% scale factor. On Windows 10 & 11, the previous GDI+ font renderer was causing too many artifacts in the console. If the stable release starts successfully but the continuous build does not, the issue might be related to this new font rendering method.

Thank you for helping us troubleshoot this, and I look forward to your feedback!

dsm, 2024-11-08:
I tried my Win 11 installation with three screens. Serial-Studio opens but does not appear on any screen, when I manually close the other screens, it appears on a single monitor and does not disappear if I turn the other screens back on, when I restart serial studio, it opens on whichever screen I closed it on, so there is a problem in the first boot, stable and continuous build has the same issue.