Serial-Studio/Serial-Studio #164: Transparent Window

jkbths, 2023-09-08:
## Describe the bug

On Windows 10 Build 19045.3324 all the windows are transparent and the UI gets unusable. 

### Screenshots

[image]

alex-spataru, 2023-09-24:
Hi! Sorry for the late reply...can you please provide me with your system's information? What graphics card(s) or drivers are you using? Did this issue appear on earlier versions of Serial Studio? 

I suspect that this issue has something to do with your graphics card driver. Serial Studio uses OpenGL to render the user interface...maybe using software rendering can help mitigate the issue.

jkbths, 2023-09-26:
This was the first time I tried Serial Studio, so I didn't use earlier versions. 
But yes I also suspect the graphic card drivers, I had the same issue with Microsoft Edge browser and I had to deactivate the use of the graphics card. Maybe it is because I run the software inside an VMware virtual machine? 
Is there a setting for using software rendering?

alex-spataru, 2023-09-26:
> Maybe it is because I run the software inside an VMware virtual machine?

Probably, I also had issues running QML-based applications on virtual machines. Note that Serial Studio is cross-platform, so it is not really necessary for you to run it in a VM (unless you have a specific reason to do so).

> Is there a setting for using software rendering?

I just added it,  please try the [latest continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous). It enables software rendering by default, you can now enable/disable software rendering in the "Settings" tab from the Setup panel.

alex-spataru, 2023-09-26:
Nice! I'll close this issue & notify you through this thread when a new stable version is released, thanks for the bug report!

closed 2023-09-26 by alex-spataru