Serial-Studio/Serial-Studio #168: Add arbitrary JSON parsing as input for dashboard

Miceuz, 2023-10-24:
**Is your feature request related to a problem? Please describe.**
I would like to use SerialStudio to visalize JSON data piping in via MQTT. I don't want to change JSON format just for SerialStudio. Now I have to change  JSON format in my applications to either be less verbose by packing CSV into JSON or use SerialStudio JSON format that would litter my model I am transmitting with a view related stuff.

**Describe the solution you'd like**
Add a communication mode 'Arbitrary JSON'. JSON file editor would allow to enter JSONPath queries into the Frame index field. SerialStudio would plot results of JSONPath queries if they are numerical. 

**Describe alternatives you've considered**
Another alternative is to wait for a first JSON packet and provide drop-downs in JSON editor populated with leafs of arbitrary JSON structure.

txf-, 2024-04-04:
I would like to add that I would really love this. Just have arbitrary json messages and then have a config that plotted the elements you needed from the message. 

Something like [MQTT Explorer does](https://mqtt-explorer.com/). 

With that you can just choose to plot any numeric value you select out of the received json messages. I would love to use Explorer, but that only works with MQTT, is abandonware and has a license incompatible with modification.

alex-spataru, 2024-11-25:
I think with the latest version this feature is possible by making use of the frame parser, please check the example here: https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#custom-frame-parsing-with-json

I'll close this issue for the moment, feel free to reopen it if the JSON parsing feature does not work for you.