Serial-Studio/Serial-Studio #172: Not receiving data in terminal

salivo, 2024-01-26:
There is no data in the Serial Studio console, but there is data in the Arduino monitor
It seams like  #86.
When i close and reopen connection it can give random data

### Screenshots
[image]
Screenshot after several clicks on connect/disconnect
[image]

## Simple code in ESP-IDF for ESP32
```c 
#include <stdio.h>

void app_main(void)
{
	while (1)
	{
		printf("*/123,123,1234/*\n");
	}
	
}
```

salivo, 2024-01-26:
### Update
these are not completely random data, I see fragments of correct data in them
[image]

salivo, 2024-01-28:
I think I found the problem, it's in RST and DRT, can I manually control it?

Sarah26-10, 2024-03-12:
I have the problem with arduino Portenta H7 Machine control. If i connect this device i receive no data at all.
What can i do?

When i connect a arduino nano every everything is working.
:)

recursivenomad, 2024-08-04:
Could this be related to #124 ([comment](https://github.com/Serial-Studio/Serial-Studio/issues/124#issuecomment-2167877139))?  I have been unable to get serial data from my Raspberry Pi Pico.

TLDR, are you on Windows and does the serial stream work if you first open it in the Arduino monitor, then restart Serial Studio?

salivo, 2024-08-05:
Yes, Thank you, I resolve the problem, just, manually disconnect DTR and RTS pin from my TTL convertor. And it works fine. I wait for update #198 with manual control DTR .

closed 2024-08-05 by salivo

alex-spataru, 2024-08-05:
Again, thanks to @recursivenomad for fixing the issue. The latest build adds a checkbox that allows users to send the DTR (Data Terminal Ready) signal upon connecting to a serial port device:

[image]

Please let me know if the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build fixes this issue to draft a new release.