Serial-Studio/Serial-Studio #175: Not working properly windows 10

ibrahimqazi, 2024-03-06:
## Describe the bug

It only opens one time just after first time installation but after that the window is stuck at the point shown in the image. Secondly, although the version is 1.1.7 but in the opening windows it still shows as 1.1.6.

### Screenshots

[image]

alex-spataru, 2024-03-10:
Hi! 

I forgot to update the splash screen design/image when I released the 1.1.7 version. Regarding the application freeze, does this issue still occur with the [continuous/beta](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) release?

Thanks,
Alex

ibrahimqazi, 2024-03-11:
> Hi!
> 
> I forgot to update the splash screen design/image when I released the 1.1.7 version. Regarding the application freeze, does this issue still occur with the [continuous/beta](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) release?
> 
> Thanks, Alex

Perfect, this version does not cause any freezing issue. Thank you

closed 2024-06-05 by alex-spataru