Serial-Studio/Serial-Studio #176: 功能请求：我非常需要对modbus协议的支持

stbanana, 2024-03-07:
**感谢**
感谢作者制作如此精美的工具！

**目前的解析方式**
目前的解析方式使用 “ start sequence " 和 ”end sequence“ 判断完整语句，这样在功能上限制是比较大的。据我所知很多通信协议是不使用“ start sequence " 和 ”end sequence“ 进行完整帧判断，而是采用 ”specified-length"  和 “CRC” 形式进行完整帧判断。

**期望添加的功能**
我所急需的是一个漂亮的modbus协议解析工具，如果作者能直接开发这个功能，我会非常高兴 : ) 。
但是从这个软件长久的发展考虑，应该添加更多自定义的解析方式， “certain bytes”  作为一个 “module" ，被赋予一些 “tag" ，不同 “tag” 有其特定的功能，让使用者自行将  “module" 自由组合为完整的解析协议，这样应该可以满足所有使用者对于协议的需求。作者进行不同标准的协议的支持也会方便许多。
我很早就有这个想法，不过受限于自身工作繁忙，我也并非桌面端软件的开发者，所以桌面软件框架需要重新学习，不得不依赖其他开发者了 : ( 。

yanyong666, 2024-03-12:
yes  i suppoese it

alex-spataru, 2024-11-17:
Hi! The latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) introduces more flexibility for handling data. You can now skip start/end delimiters entirely or choose to use only an “end sequence,” such as a line break.

Checksum support is also available, though documentation and examples are still in progress. The basic idea is to send your data as usual, appending one of the following:
- `crc8:<checksum uint8_t>`
- `crc16:<checksum uint16_t>`
- `crc32:<checksum uint32_t>`

You can find the implementations for these checksum functions [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Checksum.cpp).

For even greater customization, you can modify the [frame parser](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view) in your project to preprocess data and tailor the dashboard to your needs.

---

你好！最新的 [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) 为数据处理提供了更多灵活性。您现在可以完全跳过起始/结束分隔符，或者仅使用一个“结束序列”，例如换行符。

校验和功能也已支持，尽管文档和示例仍在完善中。基本思路是像往常一样发送数据，并在数据后附加以下之一：
- `crc8:<checksum uint8_t>`
- `crc16:<checksum uint16_t>`
- `crc32:<checksum uint32_t>`

校验和函数的实现可以在[这里](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Checksum.cpp)找到。

此外，您还可以通过修改项目中的 [frame parser](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view)，预处理数据并根据需要定制仪表盘，从而实现更高级的自定义功能。

closed 2024-11-17 by alex-spataru