Serial-Studio/Serial-Studio #180: compilation error

fu15135167211, 2024-03-19:
D:/qt5_15/Tools/mingw810_64/bin/mingw32-make -f Makefile.Release
mingw32-make[1]: Entering directory 'D:/cs/cs/Serial-Studio-1.1.7/build-Serial-Studio-Desktop_Qt_5_15_2_MinGW_64_bit-Release'
uic ../Serial-Studio-1.1.7/libs/QSimpleUpdater/src/Downloader.ui
/usr/bin/sh: D:\qt5_15\5.15.2\mingw81_64\bin\uic.exe: command not found
mingw32-make[1]: Leaving directory 'D:/cs/cs/Serial-Studio-1.1.7/build-Serial-Studio-Desktop_Qt_5_15_2_MinGW_64_bit-Release'
mingw32-make[1]: *** [Makefile.Release:14644: uic/ui_Downloader.h] Error 127
mingw32-make: *** [Makefile:45: release] Error 2
01:07:06: 进程"D:\qt5_15\Tools\mingw810_64\bin\mingw32-make.exe"退出，退出代码 2 。
Error while building/deploying project Serial-Studio (kit: Desktop Qt 5.15.2 MinGW 64-bit)
When executing step "Make"
01:07:06: Elapsed time: 00:07.

When I compile the latest  version 1.1.7 , I encountered the above fault, which indicates that uic cannot be found. I am using the Windows  os

alex-spataru, 2024-04-01:
Hi! I generally use MSVC compiler to build Qt projects on Windows, I haven't used MinGW in a while with Qt.

Another thing, based on your log it seems that you are using Qt 5.12 to compile Serial Studio. However, development of the project is currently done using Qt 6 (preferably Qt 6.5 or above).

Can you please try to compile the project using Qt6 and get back to me?

---

你好我通常使用MSVC编译器在Windows上构建Qt项目，我已经有一段时间没有在Qt上使用MinGW了。另一件事，根据您的日志，您似乎正在使用Qt 5.12来编译Serial Studio。然而，该项目的开发目前使用Qt 6（最好是Qt 6.5或以上）。你能试着用Qt6编译这个项目并回复我吗？

alex-spataru, 2024-04-01:
Another thing, when you clone the repository using Git, it is necessary to run these commands:

> git submodule init 
> git submodule update

Alternatively, you can just use:

> git clone https://github.com/Serial-Studio/Serial-Studio --recursive

Downloading the ZIP file directly from GitHub might not work, as the library dependencies (submodules) are not included in the file.

---

另一件事，当您使用 Git 克隆存储库时，需要运行以下命令：

> git submodule init
> git submodule update

或者，您可以只使用：

> git clone https://github.com/Serial-Studio/Serial-Studio --recursive

直接从 GitHub 下载 ZIP 文件可能不起作用，因为文件中不包含库依赖项（子模块）

closed 2024-06-05 by alex-spataru

lazyguoguo, 2025-01-02:
安装6.8.1 社区版，编译工具选择mingw。
在源码目录 创建 build 目录，进入build 目录运行
cmake .. -DCMAKE_RC_COMPILER="C:/Qt/Tools/mingw1310_64/bin/windres.exe" -DCMAKE_C_COMPILER="C:/Qt/Tools/mingw1310_64/bin/gcc.exe" -DCMAKE_CXX_COMPILER="C:/Qt/Tools/mingw1310_64/bin/g++.exe" -DPRODUCTION_OPTIMIZATION=ON -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="C:/Qt/6.8.1/mingw_64/lib/cmake" -G "Unix Makefiles"  创建 makefile成功

再运行  C:/Qt/Tools/mingw1310_64/bin/mingw32-make.exe
编译成功