Serial-Studio/Serial-Studio #185: How to use  Customize frame parser in Project Editor to phase my data.

maming214, 2024-04-04:
for example ,i have data such as AA BB 0A 0B F4
AA BB is frame head,F4 is this data of sum check
if frame head and check is right ,i will phase it ,or i will not phase it.
How to use  Customize frame parser in Project Editor to phase my data.
can you give me an example ,thank you

MicPon59, 2024-07-25:
Hi! Any updates on the example you were mentioning?

alex-spataru, 2024-10-13:
Hi!

I just uploaded an example project that demonstrates how to parse binary/hexadecimal data in Serial Studio: [Hexadecimal ADC Example](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/HexadecimalADC).

To run this example, please make sure to download and install the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), which allows selecting a data conversion method before passing frame data to the parser function. I'm planning to draft a new stable release around the end of October or early November.

Thank you for your patience!

closed 2024-10-31 by alex-spataru