Serial-Studio/Serial-Studio #189: macos-latest runner

dsm, 2024-05-01:
Hi, latest succesfull github action for macos-latest image is macos-12 but now runner changed to macos-14-arm64 [(available-images)](https://github.com/actions/runner-images/tree/macos-14-arm64/20240422.3?tab=readme-ov-file#available-images) . So apple silicon and intel build must be seperated.

alex-spataru, 2024-05-05:
Hi! I just finished updating the workflow file to generate both ARM64 and Intel builds for macOS. I also signed & notarized the executable to avoid any issues with newer macOS versions.

You can try the builds here: https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous

Please let me know if you find any issue, or if I should proceed to close this issue. Thanks!

dsm, 2024-05-05:
I tried my own build it also works and I'll try your build as soon as possible, thank you for your work :)

closed 2024-06-05 by alex-spataru