Serial-Studio/Serial-Studio #190: 关于窗口异常（双显示器扩展时）

QQ-X5zj, 2024-05-06:
## Describe the bug 问题描述

如果双显示器打开时，打开关于就会狂闪（主窗体靠右侧时），无法点击。窗体布局错位。错误截图如下。

### Screenshots 截图（狂闪时无法截图。主窗体稍微靠左一点，就显示关于的内容错位）

[image]

alex-spataru, 2024-05-06:
非常感谢您报告该问题。 我相信当显示器具有不同的 HiDPI 比例因子时可能会发生这种情况。 我会尝试重现该问题并及时向您通报最新情况。

谢谢，
Alex

alex-spataru, 2024-10-31:
Hi! Could you check if this issue still occurs with the latest continuous build? You can download it [here](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous).

Let me know if this resolves the problem!

你好！你能否检查一下在最新的持续构建版本中这个问题是否仍然存在？你可以在[这里](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)下载。

请告诉我是否解决了问题！

closed 2024-10-31 by alex-spataru