Serial-Studio/Serial-Studio #193: The highest supported baud rate

ZhiyuanYuanNJ, 2024-06-04:
Hi, 

When I was using a 15Mbps serial port, I found that I couldn't send or receive. Is the current maximum serial port baud rate supported 2Mbps?

Thank you :)

alex-spataru, 2024-06-04:
Hi! The maximum supported baud rate largely depends on the hardware you're using, including both the microcontroller and your computer. In my experience, baud rates higher than 115,200 are rarely necessary and can often lead to issues. For higher data transfer rates, you might want to consider other transmission methods such as UDP, network ports, or BLE, which could be more reliable for your needs.

Using very high baud rates can result in data loss, often due to issues like induced currents in the serial/USB cable, especially when dealing with long cables or noisy environments. 

Additionally, the oscillator/crystal used for setting the clock of your microcontroller can also dictate the maximum supported baud rate. 

The theoretical baud rate limit supported by Qt & Serial Studio is the maximum value of an int32_t number (2,147,483,647). You can find more details in the following links:
- [Serial.cpp, line 413](https://github.com/Serial-Studio/Serial-Studio/blob/91599fa313bd27841c22b61d768ff0d475b0df03/src/IO/Drivers/Serial.cpp#L413)
- [Qt Serial Port baud rate property](https://doc.qt.io/qt-6/qserialport.html#baudRate-prop)

ZhiyuanYuanNJ, 2024-06-05:
Hi,
After my attempts, I have made a new discovery. I found that if I manually enter the baud rate in the drop-down box, it will cause garbled characters. It should be that the baud rates at both ends are inconsistent. Is the logic in the code to directly obtain the value of the drop-down box or to obtain it according to the index?

alex-spataru, 2024-06-05:
> It should be that the baud rates at both ends are inconsistent. 

Most probably, does using any other software than Serial Studio result in the same behavior? If so, the issue probably lies on the microcontroller code or hardware.

> Is the logic in the code to directly obtain the value of the drop-down box or to obtain it according to the index?

Serial Studio converts the string displayed in the combobox to an `int32_t` number, which is then used to set the baud rate. This approach allows Serial Studio to use both the predefined baud rates and custom user-defined baud rates.

I will revalidate the string-to-integer conversion code, as the issue might be caused by a bad conversion or integer overflow.

alex-spataru, 2024-06-05:
By the way, which microcontroller are you using? Are you transmitting any non-ASCII characters, such as Chinese characters or letters with accents?

ZhiyuanYuanNJ, 2024-06-11:
Hi, 
Sorry for replying only now.

> By the way, which microcontroller are you using? Are you transmitting any non-ASCII characters, such as Chinese characters or letters with accents?

The USB serial port chip I am currently using is CH347, which supports up to 9Mbps and is a mature product. I have not found any issues with other serial port software. The characters sent are ASCII encoded, so it should still be a baud rate issue because 1000000 and 2000000 have no garbled characters, while 150000 has garbled characters.

closed 2025-06-16 by alex-spataru