Serial-Studio/Serial-Studio #196: 窗口无法最大化

Ahoge1993, 2024-07-13:
我的分辨率是1920*1080，缩放比例150%，按窗口最大化按钮，窗口会抖一下，然后恢复原状，无法最大化

alex-spataru, 2024-08-12:
Thank you for your bug report. I’ve encountered similar issues when running Serial Studio on Linux. The root cause lies in my decision to render all window controls directly within the application, rather than relying on the operating system to handle this.

I’m currently focused on a much-needed UI redesign and modernization of the project, including a transition from QMake to CMake and implementing platform-specific code to customize the window caption color to align with the application theme. However, it may take a few days to push these updates to the repository, as I want to ensure that the continuous build is fully functional before releasing any changes to the master branch.

By the way, below is a screenshot of the new UI. I'll let you know when I push the changes so that you can test if your issue is fixed. 😄 

---

感谢您提交的错误报告。我在 Linux 上运行 Serial Studio 时也遇到了类似的问题。根本原因在于我决定在应用程序内部直接渲染所有窗口控件，而不是依赖操作系统来处理这些控件。

目前，我正在专注于项目的用户界面重新设计和现代化工作，包括从 QMake 迁移到 CMake，以及实现特定平台的代码，以便自定义窗口标题栏的颜色，使其与应用程序主题保持一致。不过，可能需要几天时间才能将这些更新推送到代码库中，因为我希望在发布任何更改到主分支之前，确保持续构建功能完全正常。

顺便说一下，下面是新用户界面的截图。当我推送更改时，我会通知您，这样您就可以测试您的问题是否已修复。😄

请注意，此中文翻译是使用软件生成的。我更倾向于接收英文的错误报告（如果可能的话），或者至少在中文文本后附上英文翻译，以避免任何沟通错误。

[image]

alex-spataru, 2024-10-31:
Hi! Just to let you know that I’ve created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0) that addresses this issue along with several others. I’ll close this issue for now, but please feel free to re-open it if the problem persists with the latest release.

---

嗨！通知你我已经创建了一个[新版本](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0)，解决了此问题以及其他一些问题。我将暂时关闭此问题，如果在最新版本中问题仍然存在，请随时重新打开。

closed 2024-10-31 by alex-spataru