Serial-Studio/Serial-Studio #197: [FIX AVAILABLE] - Some microcontrollers require DTR on Windows

recursivenomad, 2024-08-05:
## Describe the bug

Serial data is not received by Serial Studio on Windows when sent from certain microcontrollers, even though the port is successfully opened.

- #124 
- #135 
- #98 
- #172 

## The fix:

This appears to be due to Data Terminal Ready (DTR) not being set.  Adding the one line of code below solves the issue for me:

```diff
// src/IO/Divers/Serial.cpp

//...
    // Open device
    if (port()->open(mode))
    {
      connect(port(), &QIODevice::readyRead, this,
              &IO::Drivers::Serial::onReadyRead);
+     port()->setDataTerminalReady(true);
      return true;
    }
//...
```

I arrived at this conclusion by first confirming QT's Serial Port example has the same issue, and found [this](https://forum.qt.io/topic/140192/communication-between-raspberry-pico-and-qtserialport-not-working-on-windows) forum post suggesting the issue was DTR.

I know nothing about working with QT, nor the inner workings of serial communication, and as such I do not know if this solution would be unsafe; but what I do know is that this one line of code fixes the problem for me with my Raspberry Pi Pico on Windows.

Things to consider from my outside perspective:
- Does the DTR flag need to be set to false upon closing?
- Should this behaviour be Windows-specific?
- Should this be hard-coded in, or an additional option in "Flow control"?

## Upload minimal *~~Arduino~~* ***Raspberry Pi Pico*** program to replicate your issue

On a Raspberry Pi Pico, build and upload the UF2 of:

```cpp
// main.c

#include <stdio.h>
#include "pico/stdlib.h"

#define PIN_LED 25

int main() {

  stdio_init_all();

  gpio_init(PIN_LED);
  gpio_set_dir(PIN_LED, GPIO_OUT);

  while (1) {

    printf("Hello, world!\n");
    gpio_put(PIN_LED, true);
    sleep_ms(100);
    gpio_put(PIN_LED, false);
    sleep_ms(100);
    
  }

}
```

```CMake
# CMakeLists.txt

# This is the version Raspberry Pi uses, no explicit reasoning provided
cmake_minimum_required(VERSION 3.13)

# Pre-initialize the Pico SDK
include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

# Set PROJECT_NAME
project(hello-world)

# Invoke the Pico SDK's CMakeLists.txt to import build settings and definitions
pico_sdk_init()

# Tell CMake what/where our executable source file is
add_executable(${PROJECT_NAME} main.c)

# Link the project to pico_stdlib as defined by the SDK
target_link_libraries(${PROJECT_NAME} pico_stdlib)

# Enable UART stdio output
pico_enable_stdio_usb(${PROJECT_NAME} ON)

# Create map/bin/hex/uf2 files, in addition to ELF
pico_add_extra_outputs(${PROJECT_NAME})
```

See [here](https://gitlab.com/recursivenomad/capacitance-measurement-module/-/blob/main/BUILDING.md?ref_type=heads) if you need a (messy) step-by-step example to setting up a Raspberry Pi Pico build environment on Windows.  
*(You can probably much more easily set up a build environment with Raspberry Pi's Pico extension for VSCode)*

Serial Studio settings:
.|.
-|-
Communication mode: | No parsing
Data source: | Serial port
Baud rate: | `115200`
Auto-reconnect: | `[ ]`
Data bits: | `8`
Parity: | `None`
Stop bits: | `1`
Flow control: | `None`

alex-spataru, 2024-08-05:
Thank you very much for identifying the issue and providing a possible fix! I don’t use Windows frequently, and primarily run it on a virtual machine, which limits my ability to test hardware natively.

Regarding your specific points:

> Does the DTR flag need to be set to false upon closing?

I reviewed the [QSerialPort](https://github.com/qt/qtserialport) module, and it doesn't specify whether `port()->setDataTerminalReady(false);` needs to be called when closing the port. We would need to test with real hardware to determine if it is necessary.

> Should this behaviour be Windows-specific?

Since QSerialPort supports setting DTR on both Windows and UNIX-based systems, I believe the software should maintain consistent behavior across all operating systems.

> Should this be hard-coded in, or an additional option in "Flow control"?

I believe the best approach would be to add a "DTR" checkbox in the GUI or include it in the "Flow control" options, allowing users to enable or disable it based on their specific use case. Using DTR might not be suitable for all microcontrollers, so providing an option would offer the necessary flexibility.

Again, thank you for your contribution. If you want, please make a pull request with your changes (so you can appear in the "contributors" section of Serial Studio), and I will later adapt the code to be able to control DTR behavior from the GUI.

recursivenomad, 2024-08-05:
PR submitted 🙂

In the commit message, I have listed that this would close #135 and #197 (this issue).  I have not directly closed #124, #98, or #172 as they contain variables beyond what I have access to, but I still referenced them to leave a breadcrumb trail.

Another thought to consider as you massage this code:
- I simply placed this flag at the end of the `open()` function.  It might function better at another location.

Thanks for accepting my contribution! ✨

alex-spataru, 2024-08-05:
Thank you very much for the PR, just added a GUI checkbox that allows users to toggle sending the DTR signal:

[image]

Please let me know if the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build works as expected!

recursivenomad, 2024-08-05:
Wow!  Thanks for so quickly putting in the energy to add the full feature 😊

I can confirm this is incredibly stable in my environment!  Transmissions are only received when `Send DTR Signal` is checked, regardless of its initial state upon connecting.

I will note that I am receiving a spurious empty newline ~5% of the time that I do not experience in other serial terminal software:

[image]

But I saw the same behaviour following my one-line PR, so this hiccup was not introduced in your commit.  I assume it is an unrelated issue (but wanted to bring it up just in case) 🙂

recursivenomad, 2024-08-05:
Update to above:

Spurious newline might be a signal integrity issue; perhaps other software is doing some filtering?  Only experiencing this when sending UART through my debugger, *not* when my Pico is transmitting via USB directly.

I'll add edits to this comment as I tinker with it over the next 15 mins or so...

alex-spataru, 2024-08-06:
> Spurious newline might be a signal integrity issue; perhaps other software is doing some filtering? Only experiencing this when sending UART through my debugger, not when my Pico is transmitting via USB directly.

I think this has more to do on how frequently the timestamp is added. I remember that I based my implementation on how the Arduino serial monitor adds the timestamp, however it is not perfect. Does the issue still appear when you disable showing the timestamp on the console?

By the way, here is the implementation of the function that adds the timestamp: https://github.com/Serial-Studio/Serial-Studio/blob/7c2ff3f8f76cb3a53c461904760628981f013ec7/src/IO/Console.cpp#L482C1-L543C2

recursivenomad, 2024-08-06:
Affirmative, issue persists regardless of timestamp, autoscroll, and new line settings:

[image]

**But** this got me thinking I'd look at the hexadecimal output!  The output seems to be parsed in specific "frames" (again, out of my depth here, but roll with me), and my microcontroller appears to send CR + LF (`0x0D 0x0A`).  I'm wondering if perhaps when CR + LF are parsed together, they are considered a single newline, but if the "frame" splits the CR (`0x0D`) and LF character (`0x0A`), then they get parsed as 2 newlines?

[image]