Serial-Studio/Serial-Studio #199: Receiving CRLF produces occasional spurious empty line in certain environments

recursivenomad, 2024-08-06:
Copied from #197 ([comment](https://github.com/Serial-Studio/Serial-Studio/issues/197#issuecomment-2271045134)):

> I am receiving a spurious empty newline ~5% of the time that I do not experience in other serial terminal software:
> 
> [image]

> Only experiencing this when sending UART through my debugger, not when my Pico is transmitting via USB directly.

> - I do not see this behaviour in PuTTY, Tabby Terminal, or the Arduino 1.x built-in serial monitor.
> - I do repeatedly see this behaviour when switching back to Serial Studio.
> - I consistently only experience this behaviour in Serial Studio when going through my debug probe (a second Raspberry Pi Pico running the "picoprobe"/"debugprobe" firmware) via UART, and I consistently do not see it when directly connected via USB to the target Pico.

> Only thing I can think of is that my debug probe's clock (in this case simply functioning as a simple UART->USB converter) is decoupled from my target's clock, and perhaps that introduces some sort of noise that Serial Studio (or maybe QT's module) uniquely produces an extra newline from?

> issue persists regardless of timestamp, autoscroll, and new line settings

> The output seems to be parsed in specific "frames" (again, out of my depth here, but roll with me), and my microcontroller appears to send CR + LF (0x0D 0x0A). I'm wondering if perhaps when CR + LF are parsed together, they are considered a single newline, but if the "frame" splits the CR (0x0D) and LF character (0x0A), then they get parsed as 2 newlines?
>
> [image]

> I should explicitly note that changing the new line setting to "NL + CR" has no effect on this (and interestingly, none of the new line settings change the Plain text output behaviour)

> Removing CR from my microcontroller output so a newline is only one character does not produce the issue.

recursivenomad, 2024-08-06:
I don't yet understand where these "frame" separations come from when UART is translated through my debug probe, but in reviewing the code [you linked](https://github.com/Serial-Studio/Serial-Studio/issues/197#issuecomment-2270364665) in the original issue, maybe I don't need to understand that - Approached from a different angle, [this line](https://github.com/Serial-Studio/Serial-Studio/blob/7c2ff3f8f76cb3a53c461904760628981f013ec7/src/IO/Console.cpp#L491C1-L491C35) seems to be the culprit when the serial stream is chopped between the `\r` and `\n`.

Maybe something like this would work?

```diff
  // Only use \n as line separator
  auto data = string;
- data = data.replace("\r\n", "\n");
- data = data.replace("\r", "\n");
+ data = data.remove("\r");
```

Simply drops all `\r` characters, leaving only `\n` in its wake.  This might not even generate any problems, as I don't believe there are any modern systems that use `\r` in the absence of `\n`?  Though I'll admit it would be nicer if the lone `\r` could still be supported.

recursivenomad, 2024-08-06:
Confirmed that the above change resolves the issue on my end, submitting a PR for consideration.

thestumbler, 2024-08-06:
> don't believe there are any modern systems that use `\r` in the absence of `\n`?

Well Windows stores text files with CR/LF line endings. Not super relevant, but I could see how that might influence the line ending ones MCU generates. For me, I always use the CR/LF when sending from an embedded application. The main reason is that sometimes you purposefully want to do a CR without a LF, for example, making a poor man’s progress bar. This wouldn’t relate to telemetry specifically, but it would be a pain to keep toggling the end of line settings in your terminal emulator. And if you were actually using a terminal 1980s style, the you definitely want both.

recursivenomad, 2024-08-06:
> The main reason is that sometimes you purposefully want to do a CR without a LF, for example, making a poor man’s progress bar.

Great point.  I've never done this, but am interested in seeing if it could be easily supported.

Could you describe what specific behaviour you'd expect when passing a CR without LF to a terminal?

- Current behaviour before the linked PR is to replace lone CR with LF in the terminal
- Behaviour after the PR is to drop all CR characters
- I'm also experimenting with a switch case based on the "Line ending" setting in Serial Studio's UI
- Another possibility could be adding a simple "Ignore incoming CR" toggle

recursivenomad, 2024-08-06:
PR [updated](https://github.com/Serial-Studio/Serial-Studio/pull/200/commits/14b1f5a51e2343afbb63870e9a18ad56576092de) with a switch case, but it feels uncomfortable for the incoming parsing to be coupled to the outgoing line ending format...  Also that implementation does not preserve the use case mentioned above.

I think fixing from this angle is beyond the scope of my familiarity with QT.  If you want to consider this angle, feel free to modify my PR accordingly...  But I think the "real" solution to this issue will be understanding why `/r` occasionally gets separated to a different timestamp from `/n`.  If that could be resolved, then something like #200 would not be necessary.

alex-spataru, 2024-08-06:
> The output seems to be parsed in specific "frames" (again, out of my depth here, but roll with me), and my microcontroller appears to send CR + LF (0x0D 0x0A). I'm wondering if perhaps when CR + LF [...]

You are correct in more than one way haha. I'll try to write a textual backtrace on how Serial Studio handles data to identify the issue:

### Part 1) General Data Flow from Microcontroller to Dashboard Frames

TL;DR: You can skip this part, but it might be useful for future reference, especially if I get myself to update the wiki or create a video tutorial.

Serial Studio processes incoming data in frames, defined by start and end delimiters set in the project editor (default /* and */, but customizable). For example, with $ as the start delimiter and ; as the end delimiter, the text:

```
$a,b,c;$d,e,f;
```

is parsed into two frames by the [`IO::Manager`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Manager.h) class:

```
Frame 1: a,b,c
Frame 2: d,e,f
```

Key functions involved are:
- [`IO::Manager::onDataReceived()`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Manager.cpp#L532), triggered by raw data from any source (serial port, BLE, TCP/UDP, etc.).
- [`IO::Manager::readFrames()`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Manager.cpp#L438), which separates incoming data into frames.

These frames are stored and processed by the [`JSON::Generator`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/JSON/Generator.h) class, replacing the "value" field in the JSON project file with actual data values. This JSON data populates the dashboard widgets.

The following architecture diagram illustrates this data flow from microcontroller to dashboard:

[image]

The diagram above is way easier to understand by imagining the data flowing from the microcontrollers (left) to the dashboard widgets (right).

### Part 2) Console Widget Context

For the console widget, raw data received by `IO::Manager` is directly processed. Unlike the structured data flow to the dashboard, we want to display raw data for debugging purposes. The console registers a slot (callback function) to handle data when the `dataReceived` signal is emitted by `IO::Manager`. This signal contains the raw, unprocessed data received.

When the `dataReceived` signal is emitted, the [`onDataReceived()`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Console.cpp#L559) slot is called, which in turn calls the [`append()`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Console.cpp#L482) function. The `append` function splits incoming data into strings using the newline (`\n`) character as a separator, adds a timestamp to each string, and displays it in the console.

It seems unlikely that a stray CR character evades the filter, implemented with this code:

```cpp
// Only use \n as line separator
auto data = string;
data = data.replace("\r\n", "\n");
data = data.replace("\r", "\n");
```

However, if `\r\n` is received in separate intervals or function calls, it might result in an extra newline being added to the console (e.g. `\n\n` , thus resulting in a timestamp being with displayed with an empty line).

In other words, we need to find a way to still separate data for this case:

```cpp
dataReceived("a,b,c,d\r");

// And some milliseconds later...
dataReceived("\n");
```

To confirm, diagnose and fix this issue, I believe we need to create a reproducible test case. This could involve a simple microcontroller program or Python script sending problematic data to Serial Studio at different intervals. Using this setup with a debugger will help us understand why Serial Studio fails to handle `\r` correctly before adding it to the console buffer.

> I should explicitly note that changing the newline setting to "NL + CR" has no effect on this (and interestingly, none of the newline settings change the Plain text output behavior)

The newline setting affects data sent from the computer to the microcontroller. For instance, pressing enter can send data as-is, or append LF, CR, or NL+CR.

> Simply drops all \r characters, leaving only \n in its wake. This might not even generate any problems, as I don't believe there are any modern systems that use \r in the absence of \n? Though I'll admit it would be nicer if the lone \r could still be supported.

I agree with @thestumbler, this depends on the system. Some systems still use CR to represent an actual line break. In such cases, removing CR characters would result in a continuous line being displayed in the console. Supporting CR characters alongside LF would indeed be the ideal solution.

alex-spataru, 2024-08-07:
By the way, I will review the PR later tonight and continue the discussion, still got some work left to do for today :(

alex-spataru, 2024-08-07:
Another thing, maybe we should add an additional condition to the filter:

```cpp
auto data = string;
data = data.replace("\r\n", "\n");
data = data.replace("\n\r", "\n"); // Not using this part would result in \n\r being converted to \n\n
data = data.replace("\r", "\n");
```

recursivenomad, 2024-08-07:
Alex, you are a wealth of knowledge!  Thank you for so clearly and thoroughly explaining the inner workings of your project.  All that you have explained corroborates what I have slowly been uncovering as I familiarize myself with your codebase / QT.

As you were writing the above message, I have been tinkering with an idea of adding a persistent flag within `IO::Console` to the `IO::Console::append` function which tracks if the final character of the previous payload was `\r`, and then handles the subsequent payload accordingly.

> To confirm, diagnose and fix this issue, I believe we need to create a reproducible test case.

As I'm almost done with composing the above thought, I think I will confirm it works and push it to my PR, and then explore properly documenting a reproducible case afterwards.

Feel free to hold off on reviewing the PR until I push an update (will ping this thread from the PR once updated).

> The newline setting affects data sent from the computer to the microcontroller. For instance, pressing enter can send data as-is, or append LF, CR, or NL+CR.

Thank you - this is something I learned once I started hacking `lineEnding()` into a switch case of my PR.  I think something which confused me is that the line ending setting is next to other settings which describe receiving data (Show timestamp, Plain text/Hexadecimal dropdown, etc).  A bit of a side-track, but perhaps it would telegraph its behaviour more effectively if the line ending selector was adjacent to the `HEX` toggle box, inline with the `Send data to device...` field?

> Supporting CR characters alongside LF would indeed be the ideal solution.

Agreed!

> maybe we should add an additional condition to the filter

Yes, that makes sense to my head as well.  Unconventional, but might as well support it!

alex-spataru, 2024-08-07:
> Alex, you are a wealth of knowledge! Thank you for so clearly and thoroughly explaining the inner workings of your project. All that you have explained corroborates what I have slowly been uncovering as I familiarize myself with your codebase / QT.

Thank you very much! Your initiative and effort in finding and addressing issues are truly appreciated. Contributions like yours are vital for keeping open-source projects alive. It can be challenging to stay motivated when most interactions are just issue reports expecting free fixes from developers. Despite our commitment to our projects, we still need to cover rent, living costs, and in general try to maintain a balanced life haha.

> As you were writing the above message, I have been tinkering with an idea of adding a persistent flag within IO::Console to the IO::Console::append function which tracks if the final character of the previous payload was \r, and then handles the subsequent payload accordingly.

I can confirm that your solution works perfectly and is elegant. I've already merged your changes. I created a small Arduino program to replicate the issue you described:

```cpp
void setup() {
  Serial.begin(115200);
  pinMode(A0, INPUT);
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
  pinMode(A6, INPUT);

  while (!Serial);
}

void loop() {
  Serial.print('/');
  Serial.print('*');
  Serial.print(analogRead(A0));
  Serial.print(',');
  Serial.print(analogRead(A1));
  Serial.print(',');
  Serial.print(analogRead(A2));
  Serial.print(',');
  Serial.print(analogRead(A3));
  Serial.print(',');
  Serial.print(analogRead(A4));
  Serial.print(',');
  Serial.print(analogRead(A5));
  Serial.print(',');
  Serial.print(analogRead(A6));
  Serial.print('*');
  Serial.print('/');
  Serial.print('\r');
  Serial.print('\n');
}
```

With your changes, no new lines were added incorrectly. The only case where new lines might be added incorrectly is when using the unconventional `\n\r` combination. However, addressing this edge case could potentially break situations where users explicitly send two consecutive line breaks or carriage returns. Finding an all-encompassing solution would likely overcomplicate the code and lead to unexpected issues.

> A bit of a side-track, but perhaps it would telegraph its behaviour more effectively if the line ending selector was adjacent to the HEX toggle box, inline with the Send data to device... field?

I agree. I made a slight adjustment to the GUI to clarify the purpose of that combobox. Here is a screenshot of the new layout:

[image]

I’ll leave this issue open for a while for anyone else to report if they encounter any problems with the changes, at least until I draft a new release. Again, thank you very much!

thestumbler, 2024-08-07:
> Great point. I've never done this, but am interested in seeing if it could be easily supported.
> 
> Could you describe what specific behaviour you'd expect when passing a CR without LF to a terminal?

I’m not sure what to expect here, because while I mentioned some reasons a CR-only ending might exist, I’m thinking of SS as a telemetry plotting display and not a general purpose terminal emulator. Having the console show the incoming telemetry, one frame per line (or I suppose multiple lines per frame if the separator wasn’t an EOL combination?) would be my primary consideration. 

If you did nothing, with CR only you’d overwrite each like and the screen would never scroll, and with LF-only the frames would walk down the page without ever going back to column 1 (assuming no modulo-column-width is being applied). From that perspective, I’d say any combination of the CR/LF or individually CR or LF should behave you’d expect — cursor moves to column 1 and advances to the next row. 

I can think of any normal reason why you’d want to you’d want to preserve the raw EOL inside any saved logged files, either. But maybe there are some very edge cases I’m missing. Furthermore I can’t think of any reason when someone would want one of CR/LF to be the frame delimiter and expect the other (LF/CR) to be meaningful data belonging to the next frame. 

This would suggest that you accept any of CR, LF, CR/LF or LF/CR as EOL, on the console do whatever you need to make EOL behave like CR/LF, and maybe offer an EOL option for logging files.

recursivenomad, 2024-08-08:
@thestumbler Thank you for your thorough perspective; I am working on an idea that I'd appreciate your input on.  Give me a few minutes to compose it below.

> and maybe offer an EOL option for logging files.

I like this idea.  This would likely be another discussion entirely, but perhaps there should be a way to "export raw" what was received from the microcontroller (or other device), alongside the current behaviour of exporting a log of what was rendered to the terminal.

recursivenomad, 2024-08-08:
@alex-spataru - I can see a possible bug from this line:

```cpp
  data = data.replace(QStringLiteral("\n\r"), QStringLiteral("\n"));
```

If someone is typically sending CRLF newlines, but then sends a lone `\r` after a CRLF pair, that extra `\r` might be dropped:

```cpp
  // Only use \n as line separator for rendering                     // Hello world!\r\n\r
  data = data.replace(QStringLiteral("\r\n"), QStringLiteral("\n")); // Hello world!\n\r
  data = data.replace(QStringLiteral("\n\r"), QStringLiteral("\n")); // Hello world!\n
  data = data.replace(QStringLiteral("\r"), QStringLiteral("\n"));   // Hello world!\n
```

I believe this behaviour is unintended... But also it could be beneficial.  I can see a solution going one of three ways:

## 1:

Reverting the EOL conversion to only include `\r\n` and `\r`, and render `\n\r` as two separate newlines:

```cpp
  // Only use \n as line separator for rendering                     // Hello world!\r\n\r
  data = data.replace(QStringLiteral("\r\n"), QStringLiteral("\n")); // Hello world!\n\r
  data = data.replace(QStringLiteral("\r"), QStringLiteral("\n"));   // Hello world!\n\n
```

## 2:

Manually accounting for these edge cases.  I have tried some ideas, but as you say above:

> Finding an all-encompassing solution would likely overcomplicate the code and lead to unexpected issues.

And in line with that statement, each manual solution I create seems to lead to more edge cases to account for.

## 3:

If the goal is to treat CR and LF "close" to a how a classic terminal would (ie. CR isn't supposed to produce a new line), then perhaps...

```cpp
  // Condense repeated/redundant \r into singular \r                    // Hello world!\r\n\r\r
  data = data.replace(QRegularExpression("\r+"), QStringLiteral("\r")); // Hello world!\r\n\r
  // Drop \r when joined by neighbouring \n
  data = data.replace(QStringLiteral("\r\n"), QStringLiteral("\n"));    // Hello world!\n\r
  data = data.replace(QStringLiteral("\n\r"), QStringLiteral("\n"));    // Hello world!\n
  // Any remaining solitary \r is treated as \n for legibility
  data = data.replace(QStringLiteral("\r"), QStringLiteral("\n"));      // Hello world!\n
```

*(If we take this approach, I would also want to account for both `\r` or `\n` spanning between payloads)*

This would be a compromise between CR accuracy and rendered legibility:
- LF always begins a new line (and for rendering legibility, it is always returned to column 1 regardless of CR presence)
- CR returns to column 1 only if necessary (if already progressing to a new line at column 1, this does nothing; if there are already characters on the current line, a new line is inserted for legibility)

@thestumbler - Given your helpful understanding above, how does the behaviour of these two bullet points sound to you?

thestumbler, 2024-08-08:
My 2 cents.... I see the console display as independent from how SS identifies frames. When you speak of a "console" or a "terminal", traditionally it was those devices that handled the EOL translation (either by DIP switches or a setup screen). That concept also fits well in this application as described in the diagram above, where the terminal emulator (bare-bones or VT100) is in the CONSOLE box.

I would suggest attaching these EOL options to the CONSOLE module only. In a real terminal, you would also be able to specify what was sent when the ENTER key was pressed and how it responded to incoming CR and LF. For example, from the VT-100 users manual Aug 1978, 1st Edition:

> _NEW LINE
The new line feature enables the RETURN key on the terminal to function like the RETURN key on an electric typewriter. When the new line feature is enabled, pressing the RETURN key generates the carriage re- turn (CR) and line feed (LF) codes. When a line feed code is received, the code is interpreted as a carriage return and line feed. When the new line feature is disabled, the RETURN key generates only the CR code; an LF code causes the terminal to perform a line feed only._

As for the frame delimiter, I see that as a completely separate issue and it is already possible (I believe) to specify that independently.

thestumbler, 2024-08-08:
Kind of related, I see from #166 that the VT-100 support is barebones and incomplete. I wonder if an existing module such as [qterminal](https://github.com/lxqt/qterminal) might be easier to incorporate?

thestumbler, 2024-08-09:
> I suppose it can be worth considering the ability to overwrite previously transmitted data for true implementation of CR functionality, but that seems beyond the scope of what Serial Studio's output terminal should render, in my opinion.

I think that use of CR-only as a true carriage return (return to col 1) EOL is going to be rare, and using CR as an EOL extremely rare, if ever. For this reason, I don’t think that getting “correct” lone-CR behavior is terribly vital. 

That said, I see no reason to make CR or LF behave different from the decades-old standards and expectations. 

I haven’t worked with QT in a couple of years (and then it was mostly Python) so I’m not familiar with the intricacies of cursor movements are made available via the APIs. If doing a true CR is difficult then I can appreciate the proposal. Otherwise, if you can easily do a CR (and a LF) in the QT terminal, then I would advise that. 

The reason I looked around for a vt100 module…  is that should there exist such a library, then you can ignore this issue entirely within the SS application itself, “passing the buck” to the vt100 module. And maybe this functionality already exists in QT itself. I am imagining a serial terminal module ( a class provided by a third party library ) that implements a terminal, and has configuration functions that let you set it up similar to how the DIP switches and factory settings menus of a real terminal permit. 

I think the EOL usage of the various CR LF combinations are now well understood and handled. As for intentional use of a lone CR or LF … I can only think of a few reasons to do this, many of them seem unimportant or unlikely to be found in setups using SS

(1) some kind of fancy banner / ASCII art kind of splash screen, where somebody might use them to shorten the length of the string constant (like sending `LF + CTRL-H+|`instead of `CR+LF+78*SPACE` to draw right side box borders. In this case I think the lone LF would be more useful than a line CR. This would fall in both unlikely and unusual categories IMHO

(2) Used somehow in a debugging CLI. This one is a real stretch if the imagination, the only plausible example I can come up with where the command prompt contains a Boolean result or error code of the command. Like 

`CMD___> RELAY K12 ON<CR>`

returns:

`<CR>OK____<CR><LF>`

 resulting in:

`OK____> RELAY K12 ON`

I’d say this one falls in the very unusual but mildly important category. 

(3) Progress bar

should be self explanatory, to make something like 

`<CR> 57.4% > ********…..`

I’d say this falls in the unusual and unimportant category I’d say. But if a terminal interpreted the CR as an EOL (CR+LF), you’d get hundreds of lines of progress on the terminal, which would be annoying to the user. 

(4) Realtime DMM-like Display

For quick and dirty testing of some component of a system, such as an IMU or A/D reading, you might want to display some values continuously 

`> 18.45 V     158 pulses/s<CR>`

I’ve done this myself, recently in fact, so I wouldn’t say this is not unusual (?) but also not really important. If SS didn’t work for me in this mode, I wouldn’t be concerned, since I would be most likely be using a terminal emulator and would only be using SS for telemetry collection / display. Or if I was using SS quite a lot, I wouldn’t do this Mickey Mouse type of display and would just use an SS widget. Maybe this also falls in the less than usual and unimportant.  

Along the lines of the notion above about passing the buck to a third party library… I wonder if a more drastic approach would be even simpler. Suppose SS provided just a basic “dumb terminal” and make SS present a pseudo-serial port that another program could connect to. That way the user could just ignore the SS console and open her own serial terminal emulator for anything more advanced like needing VT-100 emulation.  I’m not sure if this is easy or even possible.

alex-spataru, 2024-09-29:
Just re-implemented the terminal widget to add much better support for VT-100 emulation. Tried using external libraries but eventually implemented my own [widget using a QQuickPaintedItem](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/UI/Widgets/Terminal.cpp). There are still some minor issues left, but in general I can connect to most telnet servers and expect a relatively smooth experience using Serial Studio. Below a video demonstration with ASCII Star Wars:

https://github.com/user-attachments/assets/9b9a0eb9-bfca-47e7-86c9-9ca42127754a

alex-spataru, 2024-09-29:
I've also introduced a third parsing mode called *QuickPlot*. In this mode, Serial Studio will automatically plot and optionally generate CSV files for any data frame that contains numerical values separated by commas, followed by an EOL character.

For example, from your microcontroller, you can simply send:

```1,2,3,4\n```
```1,2,3,4\r```
```1,2,3,4\n\r```

And Serial Studio will then automatically generate the corresponding plots for you. 

Additionally, I've improved the project editor to make working with larger projects much more convenient and straightforward. I'd love to hear your feedback on the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)! 😄

recursivenomad, 2024-09-30:
Thank you for your continued work, Alex!

I don't have an opportunity to test at the moment, but I wanted to nudge [this comment](https://github.com/Serial-Studio/Serial-Studio/issues/199#issuecomment-2275499020) from above (I know this thread has become quite verbose, so just wanted to make sure you saw this possible bug - slightly rephrased below for clarity).

> If someone is typically sending CRLF newlines, but then sends a lone \r after a CRLF pair, that extra \r (but not subsequent \r's) might be dropped:
>
> ```c
>  // Only use \n as line separator for rendering                     // Hello world!\r\n\r\r\r
>  data = data.replace(QStringLiteral("\r\n"), QStringLiteral("\n")); // Hello world!\n\r\r\r
>  data = data.replace(QStringLiteral("\n\r"), QStringLiteral("\n")); // Hello world!\n\r\r
>  data = data.replace(QStringLiteral("\r"), QStringLiteral("\n"));   // Hello world!\n\n\n
> ```

Check out the linked comment for more context / possible solutions (if you feel this is a significant enough problem).

Looking forward to playing with the new build soon™!

alex-spataru, 2024-10-01:
Hi @recursivenomad, thank you for pointing this out!

I’ll review the comment you mentioned and will update this thread with any solutions that make it into the code for review before the release. The bug should be addressed and fixed now that it’s been identified. I appreciate your help in bringing this to my attention!

PR #200 (2024-08-07): Remove carriage returns when parsing received data
Fixes #199

#### Don't merge this yet, might try a sneaky switch case first
commit 702162a9 (2024-08-07): Remove carriage returns when parsing received data

Fixes #199
commit c439e5a5 (2024-08-07): Spot trailing CR and parse next line accordingly

Fixes #199
closed 2024-11-17 by alex-spataru