Serial-Studio/Serial-Studio #205: Button widget to talk back via Serial

MicPon59, 2024-09-24:
It would be nice it there was a button widget or something serial, with full customizable code execution, in order to send data to the device when pressed

alex-spataru, 2024-09-25:
Hello @MicPon59,

I’ve just implemented this feature. To test it, please download and install the [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), and follow these steps:

1. **Open the Project Editor** and click on the "Add Action" button:

   [image]

2. **Configure the Action:** Set a title for the button, specify the command/data to be sent when the button is clicked, and optionally choose a custom icon:

   [image]

   [image]

3. **Save and Test:** Once you've saved the project, your user-defined actions will appear on a toolbar above the other widgets. Clicking any of these buttons will send the data/command as defined in the Project Editor:

   [image]

Please give it a try and let me know your feedback!  

As you might guess from the screenshots, I’ve made several large updates and revamps in the latest commits. I’m preparing a new release soon and would greatly appreciate any feedback before finalizing it. If you encounter any issues while upgrading from the previous version, please report them so I can address them before the release.

alex-spataru, 2024-09-26:
### Example Usage (Arduino Code)

This Arduino program reads data from all ADC ports (analog inputs in Arduino-lingo) and transmits the readings via the serial port. It also supports the following actions:

- Reboot the microcontroller
- Enable the pull-up resistors on all ADC pins
- Disable the pull-up resistors (reset ADC pins to standard input mode)

```cpp
#include <avr/wdt.h>

//-----------------------------------------------------------------------------
// Serial Studio actions
//-----------------------------------------------------------------------------

void rebootSystem() {
  wdt_disable();
  wdt_enable(WDTO_15MS);
  while (1) {}
}

void setPullUp() {
  pinMode(A0, INPUT_PULLUP);
  pinMode(A1, INPUT_PULLUP);
  pinMode(A2, INPUT_PULLUP);
  pinMode(A3, INPUT_PULLUP);
  pinMode(A4, INPUT_PULLUP);
  pinMode(A5, INPUT_PULLUP);
  pinMode(A6, INPUT_PULLUP);
}

// Function to set all pins back to INPUT mode
void initAdcPort() {
  pinMode(A0, INPUT);
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
  pinMode(A6, INPUT);
}

//-----------------------------------------------------------------------------
// Initialization function
//-----------------------------------------------------------------------------

void setup() {
  Serial.begin(115200);
  initAdcPort();

  while (!Serial)
    ;
}

//-----------------------------------------------------------------------------
// Loop function
//-----------------------------------------------------------------------------

void loop() {
  // Check if there is any serial input available
  if (Serial.available() > 0) {
    String command = Serial.readStringUntil('\n');
    command.trim();

    if (command == "reboot")
      rebootSystem();
    else if (command == "pull-up")
      setPullUp();
    else if (command == "reset-adc")
      initAdcPort();
  }

  // Regular data reading and transmission
  Serial.print('/');
  Serial.print('*');
  Serial.print(analogRead(A0));
  Serial.print(',');
  Serial.print(analogRead(A1));
  Serial.print(',');
  Serial.print(analogRead(A2));
  Serial.print(',');
  Serial.print(analogRead(A3));
  Serial.print(',');
  Serial.print(analogRead(A4));
  Serial.print(',');
  Serial.print(analogRead(A5));
  Serial.print(',');
  Serial.print(analogRead(A6));
  Serial.print('*');
  Serial.print('/');
  Serial.print('\n');
}
```

#### Project File 

You can download the corresponding Serial Studio project file here: [ADC Test.json](https://github.com/user-attachments/files/17139826/ADC.Test.json)

⚠️ **Note:** It’s recommended to open the JSON file using the Project Editor in Serial Studio. While manual editing is possible, it is not advised, as it can lead to compatibility issues and unsupported projects in future releases. I’ll be updating the documentation before the next release to clarify this.

#### Dashboard screenshot/video

Here’s a preview of the dashboard in action:

https://github.com/user-attachments/assets/7bc7b4c9-d316-4eeb-b81e-066fa1e8408b

MicPon59, 2024-10-01:
Works perfectly fine, thank you for the feature!

closed 2024-10-01 by MicPon59