Serial-Studio/Serial-Studio #206: QT .dll crashing after first time opening

gblough, 2024-09-27:
## Describe the bug

A clear and concise description of what the bug is.

### Screenshots

If applicable, add screenshots to help explain your problem.

## Upload your JSON map file*

This is important in order to reproduce any UI-related issues.

## Upload minimal Arduino program to replicate your issue

This is important so that I can replicate the environment and setup that causes the bug. 

An example program could be:

```cpp
void setup() {
    Serial.begin(115200);
    while (!Serial);
}

void loop() {
    Serial.println("/*1,2,3*/");
    delay(50);
}
```

If you need me to test with a specific microcontroller, please consider [donating](https://www.paypal.com/donate?hosted_button_id=XN68J47QJKYDE) so that I can buy that MCU and help you!

gblough, 2024-09-27:
If you need any other information, just let me know.  Happy to get that to you!

alex-spataru, 2024-10-31:
Hi! Does this issue still occur with the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0)? I’ve made several updates and upgraded to Qt 6.8 LTS, which may help resolve this. In previous versions, Serial Studio had a lot of problems on older systems, as the widgets and the terminal were rendered using a QPainter, which consumed significant CPU time and interfered with the event system.

The latest version is designed to be much more efficient, using GPU rendering whenever possible and only re-painting widgets/terminal when necessary. It’s also more stable—I’ve tested it on 10+ year-old laptops running Linux and Windows 10 with no freezes or hang-ups, even in highly demanding dashboard setups. I hope this release resolves your issue, please let me know how it works for you (and your user).

alex-spataru, 2024-11-17:
Closing this issue, feel free to re-open if needed.

closed 2024-11-17 by alex-spataru