Serial-Studio/Serial-Studio #207: Connect/Disconnect Button Greyed Out

CaribouFrequency, 2024-09-28:
## Describe the bug

When connecting to a serial port, the disconnect button becomes greyed out a few seconds after connection. I cannot disconnect without closing the entire program. If I disconnect the device physically, it changes to 'connect', but is still greyed out, leaving me unable to form new connections. The connection works as expected, even though I cannot properly disconnect. It seems to primarily affect RF COM ports, but it has also happened to me while using a USB COM port.

I am running Serial Studio on KDE Neon 6.0 22.04, connecting to HC-05 modules. Everything works as expected, except for the inability to sever connections without closing the entire program.

### Screenshots
[image]

## Upload your JSON map file

I am not sure what this is, or where it is located.

## Upload minimal Arduino program to replicate your issue

I have had the issue while using this code on an Arduino Uno, but I have also experienced it running some more complicated C code on an STM32 Nucleo F446RE board.

```
#include <SoftwareSerial.h>
SoftwareSerial BTserial(2, 3); // RX | TX
 
const long baudRate = 38400; 
char c=' ';
boolean NL = true;
 
void setup() 
{
    Serial.begin(9600);
    Serial.print("Sketch:   ");   Serial.println(__FILE__);
    Serial.print("Uploaded: ");   Serial.println(__DATE__);
    Serial.println(" ");
 
    BTserial.begin(baudRate);  
    Serial.print("BTserial started at "); Serial.println(baudRate);
    Serial.println(" ");
}
 
void loop()
{
 
    // Read from the Bluetooth module and send to the Arduino Serial Monitor
    if (BTserial.available())
    {
        c = BTserial.read();
        Serial.write(c);
    }
 
 
    // Read from the Serial Monitor and send to the Bluetooth module
    if (Serial.available())
    {
        c = Serial.read();
        BTserial.write(c);   
 
        // Echo the user input to the main window. The ">" character indicates the user entered text.
        if (NL) { Serial.print(">");  NL = false; }
        Serial.write(c);
        if (c==10) { NL = true; }
    }
 
}
```

alex-spataru, 2024-10-01:
Hi @CaribouFrequency, thank you for reporting this issue.

Could you please let me know if this problem still occurs with the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) release?

I have made numerous changes in the codebase since the last stable release, so creating a patch for the older version might not be practical. However, if this issue persists in the current "beta" version, I'd like to address it before finalizing the next stable release.

I’d appreciate it if you could help by testing this version, thanks!

CaribouFrequency, 2024-10-01:
Hi @alex-spataru 

The problem still occurs with the latest release. I have observed in both versions that if you change data source/device setup from serial to one of the other options, Network Socket or Bluetooth LE, and then switch back to Serial, it will disconnect and ungrey the COM port dropdown, and the connect button.

Is there any other debugging information I can provide?

alex-spataru, 2024-10-01:
Hi, the issue might originate from the Qt signals responsible for updating the connect button or the serial port list not being emitted correctly, which leaves the UI in an indefinite state. 

Could you try manually switching the serial port device list to ‘Select Port’ and then back to the desired serial port? I’d like to confirm whether this reactivates the connect button.

CaribouFrequency, 2024-10-03:
When it gets stuck in this way, the device list also becomes inoperable, so I cannot do that.

I have noticed another peculiar behaviour. When I was connected to a USB COM port, the connect button and device selection menu turned grey immediately after I typed `sudo rfcomm bind 0 <address>` into a console. Could there be a problem with the way it is obtaining the device list?

alex-spataru, 2024-10-05:
Hi, I’ve managed to replicate the issue. It occurs when the list of serial port devices changes while a device is actively connected. Specifically, I can reproduce this by connecting to a device and then physically adding or disconnecting another device from the computer. I’m currently working on a fix and will keep you updated on the progress.

alex-spataru, 2024-10-05:
I've just added a commit that should fix this issue, at least based on my current setup. Could you please check if the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) resolves the problem on your side as well? Let me know how it goes.

CaribouFrequency, 2024-10-05:
The issue appears to be resolved on my end. I cycled between all of my devices, which includes two USB COM ports, and one rfcomm. No elements of the UI locked up, and I was able to switch between them without issue.

alex-spataru, 2024-10-06:
> The issue appears to be resolved on my end. I cycled between all of my devices, including two USB COM ports and one RFCOMM. No UI elements locked up, and I was able to switch between them without any problems.

Glad to hear that! I’ll close this issue for now. If you encounter this bug again, feel free to reopen it or let me know.

closed 2024-10-06 by alex-spataru