Serial-Studio/Serial-Studio #208: Serial Studio Crashing When Attempting to Open CSV Files

zachrainnay, 2024-10-11:
I apologize for not including related files such as JSON or code, as I believe they may not be necessary for addressing this issue.
I encountered a recurring issue with Serial studio. Every time I attempt to open a CSV file to view the plotted data, the program crashes, stops responding, and then closes itself automatically. This issue happens consistently regardless of the file size. I’ve tried opening both small and large CSV files, but the result is the same—Serial studio becomes unresponsive and crashes.

Serialstudio Version: 2.1.0

alex-spataru, 2024-10-11:
Hi! Can you please share with me a CSV file that causes Serial Studio to crash?

I can't seem to replicate this issue on my end...thanks!

zachrainnay, 2024-10-11:
Thanks for getting back to me! I'm sending you two of the CSV files that I've tried. These files contain GPS coordinates, which are from a public place, so I don’t mind sharing them with you. I’ve also renamed the file names to reflect their sizes for easier reference.

In addition to the CSV files, I’m also sending my JSON project editor file, which only contains a GPS plot. For additional context, I was able to open the CSV file successfully just once after updating to the latest version. However, after that, Serial Studio stopped working with the files and has consistently crashed since then.

Please let me know if you need any more details.

[TEST_GPS.json](https://github.com/user-attachments/files/17344028/TEST_GPS.json)
[2.84 MB.csv](https://github.com/user-attachments/files/17343899/2.84.MB.csv)
[157 KB.csv](https://github.com/user-attachments/files/17343900/157.KB.csv)

alex-spataru, 2024-10-11:
I’ve identified a couple of issues in your configuration:

**1. Separator and End Delimiter Sequences Are Identical**
The separator and end delimiter sequences are currently set to the same value, which prevents Serial Studio from correctly parsing the incoming data. Could you try using distinct sequences for the start delimiter, end delimiter, and separator? This should resolve the issue.

[image]

**2. Malformed CSV Output**

Since the frames are not being parsed correctly, the resulting CSV file is malformed. Here’s an example of what the malformed CSV looks like:

```csv
Column1,Column2,Column3,Column4,Column5
0,-7.765861,110.374443,0,0.0926
0,-7.765861,110.374443,170.899994,0.0926
0,-7.765861,110.374443,170.899994,0.0926
0,-7.765861,110.374443,170.899994,0.0926
```

In this case, the first column should contain the RX Date/Time, but only received values are appearing. As a result, Serial Studio cannot calculate the time difference between frames, which in turn causes the crash.

- Please ensure that the start delimiter, end delimiter, and separator sequences are distinct from one another. This should fix the issue and should be able to generate "playable" CSV files.
- If you need more information on project parameters, you can refer to the following documentation in the wiki:  
  - [Operation Modes](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes#json-project-file-mode)  
  - [Project Editor](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor)

Once you've made the adjustments, please let me know if the issue is resolved.

To help prevent similar issues in the future, I’ll also be adding validation to:
- Ensure that the CSV file is properly formatted before it’s opened.
- Check for invalid project configurations in the Project Editor.

Thanks again for reporting this issue, and let me know if you need further help!

alex-spataru, 2024-10-11:
By the way, if you’d prefer not to change the format of your data too much, you can simply remove the last `;` from your frame and set the end delimiter to a newline character (`\n`). In most of my projects, I send each frame on a separate line, and using `\n` as the end delimiter is a perfectly valid approach.

zachrainnay, 2024-10-11:
Thank you for the detailed explanation! I've made adjustments to the delimiter and separator settings to align with my CSV file, and the dashboard now shows up successfully. However, when I tried to open a different file afterward, Serial Studio crashed again.

Regarding the "Date/Time" column, I'm unsure whether it requires both the date and time, or if it's possible to use only one. The wiki doesn't specify that the first column after the start delimiter must be date/time, so I tried modifying the first column to include a date ('`20/10/2024`'). 
The row now looks like this
`$20/10/24, GPS DATA, GPS DATA, GPS DATA, random;`
After reopening Serial Studio with this change, the dashboard displayed correctly, but once again, when I tried to open a different file—or even reloaded the same file—it crashed.

Interestingly, I noticed that the dashboard **sometimes** works if I follow these steps:

1. Create a new project file every time I want to open a CSV file.
2. If I don't create a new project file, I have to clear the console before launching another CSV file.
3. I don't have to change the first column to Date/Time as long as I did number 1 or number 2 mentioned above.
4. if none of the above works, close and wait some time then reopen Serial Studio.

Is this expected behavior, or is there another configuration I might be missing? 
Thanks again for your assistance, and I look forward to your feedback!

alex-spataru, 2024-10-11:
Don't worry about adding the date/time or sending it from your microcontroller project. Serial Studio adds this part automatically for you in the CSV file. 

You can see an example GPS implementation here: https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/TinyGPS

Regarding the crash, opening a malformed CSV file unfortunately will always crash the application (at least until I add CSV validation code), even if project settings are correct.

I'm on my cell phone, I'll get back to you with a full answer later today.

alex-spataru, 2024-10-11:
By the way, you might want to obtain some readings from your microcontroller again and check how Serial Studio formats the Date/Time column. It uses a specific format that includes date, hour, minute, second & milliseconds...just adding the date won't work.

Here is the format that Serial Studio requires in the fist column (which ideally is automatically added by Serial Studio, not by the microcontroller or manually in a Spreadsheet editor):

`yyyy/MM/dd HH:mm:ss::zzz`

Where:
- `yyyy` is the year
- `MM` is the month number
- `dd` is the day number 
- `HH` is the hour (0-11)
- `mm` is the minute
- `ss` are the seconds
- `zzz` are the milliseconds (0-999)

Remember to add leading zeroes to each item if editing manually 😄

zachrainnay, 2024-10-11:
Thank you so much for your time and support, I really appreciate it! While I wait for your full response, I have a quick question: if Serial Studio is able to open my dashboard several times but then fails to do so on subsequent attempts, does that indicate that my CSV format is correct? Since it opens successfully sometimes, I’m wondering if the issue could be something else, rather than the CSV format itself.
I’m looking forward to your insights whenever you have a moment. Thanks again for all your help!

alex-spataru, 2024-10-11:
You're welcome!

> I have a quick question: if Serial Studio is able to open my dashboard several times but then fails to do so on subsequent attempts, does that indicate that my CSV format is correct?

There might be several reasons for the failure, one might be that the state of the CSV player is not reset correctly when opening a malformed CSV file, or another could be that dashboard items are not updated correctly (eg the GPS map receives an invalid latitude/longitude combination).

To give you an exact answer I would need to launch a debugger and see what is happening internally in Serial Studio when parsing your CSV files...I would really appreciate if you could share with me your updated project file & CSV files that cause the issue.

I'll get back to you later, hope the last messages help you fix your issue!

alex-spataru, 2024-10-11:
Also, if possible, please download the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build. 

Version still is 2.1.0...but if I recall correctly, I made some minor changes in the CSV module to ensure that it works correctly when working with projects that have multiple datasets assigned to the same frame index. You might still get a crash or inconsistent behavior for the moment, but at least we'll be using the exact same version of Serial Studio to make debugging easier.

zachrainnay, 2024-10-11:
> Also, if possible, please download the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build.

Yes, I’m using the latest build. When I downloaded it, the release was marked as 3 hours ago. I also checked the link you shared, which shows it was released 9 hours ago. It seems we're both working with the same version, so we should be in sync.

alex-spataru, 2024-10-13:
Hi!

I just implemented a feature that allows you to select how to handle date/time information when dealing with an invalid CSV file. If the CSV doesn’t have a valid date/time column, the project will now prompt you to either select a different column or manually set a custom time interval between rows.

This won’t fix the parsing issues with the CSV files you shared, but it should help you work with the readings by allowing you to customize the time interval.

⚠️ Please note: Clicking on the "Select a date/time column" may cause the program to appear unresponsive with large CSV files. The program isn't actually frozen; it’s processing a large amount of data to prepare the CSV for parsing. I’ll work on a more efficient implementation of the column move operation in the future.

Additionally, I’ve attached a JSON project that should parse your CSV files without modification, provided you choose the "Set interval manually" option.

Let me know if you encounter any issues or need further adjustments!

[TEST_GPS.json](https://github.com/user-attachments/files/17353357/TEST_GPS.json)

alex-spataru, 2024-10-14:
Thanks for testing the latest build of Serial Studio and for your feedback! I'm glad to hear that the update has improved your experience with CSV files.

I looked into the issue you mentioned and found the cause. The problem happens when trying to open a new CSV file while one is currently playing. I've fixed it by disabling the "Open CSV" button while a file is playing or open. To open a new file, just close the player dialog, and the button will be available again.

As for the timestamp format, you're correct! The `yyyy/MM/dd HH:mm:ss::zzz` format is correct. Here's the full list of timestamp formats Serial Studio supports:

1. `yyyy/MM/dd HH:mm:ss::zzz`
2. `yyyy/MM/dd/ HH:mm:ss::zzz`
3. `yyyy/MM/dd HH:mm:ss`
4. `yyyy/MM/dd/ HH:mm:ss`

Any of these should work when creating CSV files.

Thanks again for your patience and for helping improve Serial Studio! If you run into any other issues or have more suggestions, feel free to reach out.

alex-spataru, 2024-10-31:
**Closing this issue for now**—feel free to let me know if it should be re-opened. By the way, I’ve just released a [new stable version](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0) that addresses this and several other issues. Please let me know if you encounter any further bugs!

closed 2024-10-31 by alex-spataru