Serial-Studio/Serial-Studio #214: How can I filter data to achieve？

yeclimeric, 2024-10-24:
Hello, I'd like to perform a custom filtering operation on the received data. Is there a current solution within the program? Can I achieve this simply by using the frame parser?

alex-spataru, 2024-10-27:
Hi! I think you can implement this on the frame parser function, I would approach this by storing a global array with the frame data and maybe implementing a simple moving average filter. 

Here is a quick, untested moving average filter outputted by ChatGPT:

```js
// Global array to store history of values for each data position
let history = [];

// Function to compute the moving average of an array
function movingAverage(arr, windowSize) {
    let sum = arr.slice(-windowSize).reduce((acc, val) => acc + val, 0);
    return sum / Math.min(arr.length, windowSize);
}

// Frame parser function with moving average filter
function parse(frame, separator) {
    // Define the window size for the moving average
    const windowSize = 3;

    // Parse the frame into individual values
    let parsedValues = frame.split(separator).map(Number);

    // Initialize history arrays if necessary
    if (history.length === 0) {
        history = Array(parsedValues.length).fill([]).map(() => []);
    }

    // Update the history and compute the moving average for each position
    let smoothedValues = parsedValues.map((value, index) => {
        // Add the new value to the history for this position
        history[index].push(value);
        
        // Compute the moving average
        let avg = movingAverage(history[index], windowSize);

        // Maintain the history size for the window size
        if (history[index].length > windowSize) {
            history[index].shift(); // Remove oldest value
        }

        return avg;
    });

    // Log for debugging purposes
    console.log("Received frame: " + frame);
    console.log("Parsed values: " + JSON.stringify(parsedValues));
    console.log("Smoothed values: " + JSON.stringify(smoothedValues));

    // Return the smoothed values
    return smoothedValues;
}
```

Let me know if this helps you, you can read more about the Frame Parser Function [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view). Also, be sure to use the [latest continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), I still haven't drafted a release with all the features that are shown in the wiki.

alex-spataru, 2024-11-17:
Closing this issue to keep things clean. Please let me know if I should re-open it.

closed 2024-11-17 by alex-spataru