Serial-Studio/Serial-Studio #215: 控制台字符显示太宽并且长数据会丢失显示

Lancer-AC, 2024-10-31:
## Describe the bug

A clear and concise description of what the bug is.

### Screenshots

If applicable, add screenshots to help explain your problem.

## Upload your JSON map file*

This is important in order to reproduce any UI-related issues.

## Upload minimal Arduino program to replicate your issue

This is important so that I can replicate the environment and setup that causes the bug. 

An example program could be:

```cpp
void setup() {
    Serial.begin(115200);
    while (!Serial);
}

void loop() {
    Serial.println("/*1,2,3*/");
    delay(50);
}
```

If you need me to test with a specific microcontroller, please consider [donating](https://www.paypal.com/donate?hosted_button_id=XN68J47QJKYDE) so that I can buy that MCU and help you!

Lancer-AC, 2024-10-31:
[image]

alex-spataru, 2024-11-01:
Hi! This feature is planned. I encountered an issue with displaying Chinese characters correctly while maintaining uniform character width. Ensuring equal width for each character preserves the monospace font style, enables accurate rendering of selection rectangles, and prevents characters from overlapping in the console widget.

You can view the code that calculates character width specifically for Chinese in Serial Studio [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/UI/Widgets/Terminal.cpp#L630). Currently, the font used is [Noto Sans Mono](https://fonts.google.com/noto/specimen/Noto+Sans+Mono), but it may be possible to use a different monospace font that better supports Chinese characters. Could you recommend any open-source or free monospace fonts optimized for Chinese?

Please see the attached screenshot for an example of how the console widget renders Chinese characters when this feature is disabled. Hopefully, this helps illustrate the reason behind enabling this feature:

[image]

---

你好！该功能已经在计划中。我在保持字符等宽显示时遇到了正确显示中文字符的问题。确保每个字符的宽度相同可以保持等宽字体样式，准确渲染选择矩形，并防止控制台小部件中的字符重叠。

您可以在Serial Studio的[此处代码](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/UI/Widgets/Terminal.cpp#L630)中查看专门为中文计算字符宽度的逻辑。目前使用的字体是[Noto Sans Mono](https://fonts.google.com/noto/specimen/Noto+Sans+Mono)，但也可能使用另一种更适合中文的等宽字体。您能推荐任何开源或免费的等宽中文字体吗？

请参见附加的截图，了解该功能被禁用时控制台小部件中中文字符的渲染效果。希望这可以帮助说明启用此功能的原因：

[image]

alex-spataru, 2024-11-01:
Another potential issue is that the [QFontMetrics](https://doc.qt.io/qt-6/qfontmetrics.html) class, which calculates font geometry, only seems to provide accurate values when calling [averageCharWidth()](https://doc.qt.io/qt-6/qfontmetrics.html#averageCharWidth) for “common” Unicode characters — as defined by European standards… 😂

You can see the implementation [here](https://github.com/radekp/qt/blob/master/src/gui/text/qfontmetrics.cpp#L424). I’ll dig into how QFontMetrics works and try to implement a solution tailored for each language, maybe this can help us fix the issue.

---

另一个潜在的问题是，[QFontMetrics](https://doc.qt.io/qt-6/qfontmetrics.html)类用于计算字体几何时，在调用[averageCharWidth()](https://doc.qt.io/qt-6/qfontmetrics.html#averageCharWidth)时似乎仅对“常见”Unicode字符（按照欧洲标准定义 😂）提供准确的值。

可以在[这里](https://github.com/radekp/qt/blob/master/src/gui/text/qfontmetrics.cpp#L424)查看实现。我会深入研究QFontMetrics的工作原理，并尝试为每种语言实现一个针对性的解决方案，这或许可以帮助我们解决这个问题。

alex-spataru, 2024-11-10:
Hi! I’ve implemented a fix for the character width issue. You can try the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) to see if it resolves the problem on your end as well.

The fix adjusts how we handle character widths in the console display. Instead of relying on a single, fixed cell width for all characters, we now dynamically calculate the width for each character as it’s rendered. This allows wider characters (like Chinese) to display properly without disrupting the overall alignment.

During development, I temporarily added grid lines to the console to help visualize and validate the new character width adjustments. Here’s how it looks with the grid lines enabled for testing:

**English data:**
[image]

**Chinese data:**
[image]

These grid lines were only used to debug and confirm the alignment adjustments—they won’t appear in the final user interface. You’ll notice that character widths now automatically adjust based on each character, ensuring a clean layout even with mixed-width characters. Here’s the final result without grid lines:

[image]

Manually rendering console data is harder than I thought 😄  — hopefully, this fix addresses all the issues! Let me know how the latest continuous build works for you to know if I should close this issue.

---

你好！我已经修复了字符宽度的问题。你可以尝试最新的[持续构建版本](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)，看看该问题是否在你那边也得到了解决。

此修复调整了我们在控制台显示中处理字符宽度的方式。我们不再依赖于所有字符的固定单元格宽度，而是现在在渲染时为每个字符动态计算宽度。这使得较宽的字符（如中文）能够正确显示，而不会影响整体对齐。

在开发过程中，我临时添加了控制台网格线，以帮助可视化和验证新的字符宽度调整。以下是启用网格线进行测试的效果：

**英文数据：**
[image]

**中文数据：**
[image]

这些网格线仅用于调试和确认对齐调整——它们不会出现在最终用户界面中。你会注意到，现在字符宽度会根据每个字符自动调整，即使是混合宽度的字符也能保持整洁的布局。以下是移除网格线后的最终效果：

[image]

手动渲染控制台数据比我想象的要难 😄 希望这个修复能解决所有问题！请试用最新的持续构建版本，让我知道问题是否解决，以便我决定是否关闭此问题。

closed 2024-11-17 by alex-spataru