Serial-Studio/Serial-Studio #218: XY Plotting

DylanBrianBenton, 2024-11-10:
Hi Mate, 

Not sure if it can be be done by customising the accelerometer widget, but is there an abillity to plot on a X,Y grid with locations?

eg -10,-10 to 10, 10 grid rendering plots like -3,2 on a widget when sent through serial?

Thanks for the great software and sorry if its possible and ive missed it.

alex-spataru, 2024-11-10:
Hi!

At the moment, the X-axis is set to load a fixed number of sample data points, so there isn’t an option to plot specific X,Y coordinates directly. However, adding this feature is something I’m definitely considering for the future—I think it could enable some really interesting plots. I’ll keep you updated in this thread on any progress.

Thanks for the suggestion, and feel free to reach out if you have other ideas.

DylanBrianBenton, 2024-11-10:
Yeah, I think that would be great. Im working on a esp32 nerf dart board and I'm trying to plot xy coordinates I'm sending. This tool would be fantastic for testing
If there's any way I can help let us know

alex-spataru, 2024-11-29:
Hi,

I’ve added initial support for this feature. In the [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), you can now select the X-Axis for each dataset with plotting enabled in the project editor:

[image]

You can choose any dataset as the X-Axis source, or stick to the default behavior by selecting samples/points as the X-Axis. Here’s an example with what I think is a particularly interesting XY plot: [**Lorenz Attractors**](https://en.wikipedia.org/wiki/Lorenz_system).

https://github.com/user-attachments/assets/c524dff2-a482-4fee-96da-ec83fd8e5aa1

Regular widget features, like individual windows, are still supported:

[image]

For now, performance and stability might take a hit while I refine the algorithms for handling X and Y data in real time. That said, I’m aiming to include this in the next release of Serial Studio (v3.0.7).

Let me know what you think!

alex-spataru, 2024-11-29:
Btw, just added an example project with Arduino code + Serial Studio project [here](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/LorenzAttractor). Please let me know how it works for you!

tarman3, 2024-11-29:
Started testing. Looks cool.
Now it is possible to show time.

alex-spataru, 2024-11-29:
@tarman3 Thank you very much for the feedback! Please let me know if you find any issue 🐞

Frenkenstein6, 2024-12-01:
What seems to work, get it hard first. then go into what you want.   ₦₦ł₭₭ł.

alex-spataru, 2024-12-02:
@Frenkenstein6 Stop posting inappropriate content. This [is](https://github.com/Serial-Studio/Serial-Studio/issues/67#issuecomment-900243882) not the [first](https://github.com/Serial-Studio/Serial-Studio/issues/216) time. This platform is for project discussions, not your childish spam. Continue, and I’ll take action. Grow up, get a life, and stop wasting everyone’s time. Thank you.

TheAlbertDev, 2025-03-16:
Hi @alex-spataru,
First of all, thanks for your amazing work. I have installed the mac version in the continuous release, however I'm not able to view this feature. Something should be enabled in the app?
Thanks again.

closed 2025-04-21 by alex-spataru