Serial-Studio/Serial-Studio #220: BLE Write / Read

DylanBrianBenton, 2024-11-12:
Sorry if this is a silly question, but I can connect to a service, but is there a way to target a specific characteristic?

Also, is there the ability to write/send to a characteristic? Eg to trigger a flag

Thanks

alex-spataru, 2024-11-13:
Hi, not a silly question at all! The oversight was actually on my part, as I didn’t add support for this initially. I implemented BLE for a quick project during college and didn’t maintain or extend the module much afterward.

However, with the [latest continuous release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), you can now select specific characteristics and descriptors to read from or write to.

A few details to note:
- If a characteristic’s descriptors include a CCCD (Client Characteristic Configuration Descriptor), Serial Studio will automatically write `0100` (in hex) to enable notifications for that characteristic, but you can also play with it manually by using the console.

Here’s a screenshot showing an example where I read the battery level of my phone:

[image]

As you can see, my phone’s battery was at 17% (`11` in hex). I decided to charge it, and the level soon went up to 18% (`12` in hex).

Hope this helps! Let me know if you have more questions.

alex-spataru, 2024-11-13:
With the latest build I also added an option to ignore start/end sequences in BLE and consider each received value as a frame...now I can build a dashboard with the battery level of my phone directly in Serial Studio 😄 

[image]

To read hexadecimal/binary bytes sent by my phone, I use the frame parser function used in the [HexadecimalADC](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/HexadecimalADC) example. I also attached the project file if you want to play with it.

[BLE Battery.json](https://github.com/user-attachments/files/17737861/BLE.Battery.json)

dsm, 2024-11-15:
We have a device using nina-b1 ble module, We use this module via our mobile phone over two years and I am writing a bootloader over ble and python script for firmware update(I use bleak) but I try your app to connect ble module using latest build with no success, it connect transparenrt serial comm characteristic and notify but there no data rx/tx and also connect and disconnect button crashed the app.

alex-spataru, 2024-11-15:
@dsm Thank you very much for your feedback! I will investigate the BLE connectivity issue further and get back to you later.

Regarding the app crash: If possible, could you use [WinDbg](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/) to load Serial Studio and provide a backtrace of the crash? Alternatively, if you’re comfortable with it, you could download Qt 6.8.0, compile the source code, and debug the issue directly using Qt Creator. Either approach would be immensely helpful in identifying and resolving the problem.

I’ve recently moved the [frame extraction code](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/FrameReader.cpp) to a separate thread. This change allows Serial Studio to handle high-frequency data more effectively while maintaining UI responsiveness (e.g., processing intervals of less than 1ms between frames). This enhancement is particularly beneficial for TCP/UDP dashboards with local sockets. While connecting to a “remote” device or using a serial port introduces inherent delays, separating the code into different threads ensures better performance overall.

I have thoroughly tested these changes on macOS and Linux, but Windows—especially the MSVC build of Serial Studio—has not been extensively tested yet. Your backtrace would provide a lot of useful information for addressing the issue. Thank you again!

dsm, 2024-11-15:
@alex-spataru No problem, I compile serial studio time to time, I have support some Qt project and we have so many issue using MSVC build so we use msys2(UCRT) to create windows build or maybe using clang-cl. For 2 weeks I don't have much time to investigate this problem but I can do my best, thanks.

alex-spataru, 2024-11-15:
Thank you very much! I’ll make sure to update this thread if I manage to replicate the issues you’ve reported. I completely agree—MSVC can be quite a pain when working on cross-platform projects. That said, it does have its benefits, as it often helps identify "subtle" issues within the codebase.

alex-spataru, 2024-11-16:
@dsm I managed to replicate the crash on Windows, it seems that it happens on the [`close()`](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/BluetoothLE.cpp#L65) function, and only from time to time. The `close()` function is also called when [opening](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/BluetoothLE.cpp#L212) a device to reset the state of the BLE module.

Specifically, the debugger indicates that the crash occurs on [this](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/BluetoothLE.cpp#L65) line:

```cpp
void IO::Drivers::BluetoothLE::close()
{
  // Clear connected flag
  m_deviceConnected = false;

  // Restore initial conditions
  m_descriptors.clear();
  m_serviceNames.clear();
  m_characteristics.clear();
  m_descriptorNames.clear();
  m_characteristicNames.clear();

  // Forget selected characteristic & descriptor
  m_selectedDescriptor = -1;
  m_selectedCharacteristic = -1;

  // Delete previous service
  if (m_service)
  {
    disconnect(m_service);
    m_service->deleteLater();
    m_service = nullptr;
  }

  // Delete previous controller
  if (m_controller)
  {
    disconnect(m_controller);
    m_controller->disconnectFromDevice();
    m_controller->deleteLater();
    m_controller = nullptr; // <- Crash occurs here...
                            // Why? I don't have a clue yet, I'm just re-assigning the pointer here...
  }

  // Update UI
  Q_EMIT servicesChanged();
  Q_EMIT descriptorsChanged();
  Q_EMIT characteristicsChanged();
  Q_EMIT deviceConnectedChanged();
}
```

alex-spataru, 2024-11-16:
And at least with my test device (my phone, don't have a BLE-enabled MCU nearby right now), the data transmission works correctly when selecting the service and characteristic that I'm interested in, I don't even need to select the CCCD descriptor or to manually send the `0100` command to the CCCD to start data transmission.

[image]

Mind you, in the screenshot above I run an `arm64` version of Serial Studio compiled with MSVC on a Windows 11 virtual machine.

I still need to find the time to install Windows on my test computer (HP Spectre x360, 2013)...for the moment it comfortably runs OpenSUSE without any overheating issues or the fans going crazy.

dsm, 2024-11-16:
I think msvc is the problem can you add this options to msvc, msvc handle differently so many thinks and it didn't ISO compliance also pointer handling different  /vmg solve this and others will also improve, add_compile_options(/permissive- /Zc:__cplusplus /Zc:preprocessor /MP /vmg)

alex-spataru, 2024-11-17:
Hi @dsm,

I’ve fixed the crash on my end. In addition to updating the compiler flags as you recommended, I found it necessary to use `delete` instead of QObject’s `deleteLater()`. Interestingly, Qt developers use `delete` in their [BLE Scanner example](https://doc.qt.io/qt-6/qtbluetooth-lowenergyscanner-example.html), so I followed suit.

Could you please test the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) to confirm if it resolves the crash on your side? Additionally, let me know if you’re now able to receive data using your NINA-B1 BLE module.

I also experimented with MinGW, but unfortunately, BLE isn’t supported on the Qt MinGW build. :(

Looking forward to hearing your feedback!

dsm, 2024-11-17:
the crash problem has gone but I can't test ble module, I'll test tomorrow.  Btw, it is not clear when we click on the buttons, it changes color when we hover over it, but it is difficult to notice very little, it can be nice if the background gets darker when pressed, like the setup and console buttons. I can also open a separate issue.

alex-spataru, 2024-11-17:
> The crash problem has gone but I can't test BLE module, I'll test tomorrow.

Glad to hear that! Thank you very much for the support and the feedback.

> Btw, it is not clear when we click on the buttons, it changes color when we hover over it, but it is difficult to notice very little, it can be nice if the background gets darker when pressed, like the setup and console buttons.

Just pushed a commit that fixes this issue, now toolbar buttons will display a dark background when clicked. See the video below for a visual demo:

https://github.com/user-attachments/assets/99221c58-ce5c-4fe2-8fb3-0892a958751a

Btw, you can also try out different themes and color schemes in the "Settings" tab, some of them are optimized depending on the environment that you are operating Serial Studio, such as "Outdoor Day" and "Outdoor Night".

dsm, 2024-11-18:
there is no data receive or transmit over BLE module. I am not sure but maybe you can look these pages to improve ble. I don't find QLowEnergyCharacteristic::Notify in code.  https://doc.qt.io/qt-6/qlowenergydescriptordata.html#isWritable and https://doc.qt.io/qt-6/qlowenergycharacteristic.html#PropertyType-enum

In python with bleak I just connect and notify gatt char there is no problem with data rx tx. Some times QLowEnergyCharacteristic::WriteNoResponse is good We should be select some option.

dsm, 2024-11-18:
Btw can you change ble device list name to includes both name and mac address and Can you add mac address filtering. Thanks

alex-spataru, 2024-11-18:
Hey @dsm,

I’ve updated the code to use QLowEnergyCharacteristic::Notify instead of QByteArray::fromHex("0100"). It’s possible that the CCCD enable data is (somehow) platform-dependent. Could you check if the latest build works for you? The CI jobs should finish in about 10-15 minutes, so please give it a try once it’s ready.

Quick question—are you selecting the characteristic from the combobox after choosing the service? It sounds dumb, but that could be the issue, haha. Previously, all characteristics with valid CCCD descriptors would automatically be set to notify. You might need to experiment with the characteristic combobox to find the one that actually sends the TX/RX data.

Also, if you can, please share your Python script. I’d like to take a look at how you’re communicating with the device.

Thanks!

alex-spataru, 2024-11-18:
I’ve updated the code to automatically report the current value when connecting to a characteristic, and to display a message box in case of any error.

Based on your setup, where TX/RX is triggered by sending data to the BLE device, here are a couple of potential scenarios to consider:
- Writing might be failing at this [line](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/BluetoothLE.cpp#L161). It might be worth investigating what is happening there.
- This might not apply to your case, but selecting a different line ending (near the “Send” text field) could also be worth trying.

Let me know if this helps! It's kind of late for me right now, but I'll check this thread in a few hours.

dsm, 2024-11-18:
After notify 703 char. become readable and writable(ble device state change) but our device only response cmds not send any data without receiving cmd. We don't use line endings. We also have bootloader to upgrade firmware over ble module so we use ble module more than 2 years, just ensure serial studio also works. I'll test app on macos today and give you feedback

dsm, 2024-11-18:
please correct me if I am wrong but normaly I write data to characteristic on python and qt example shows that line to write characteristic and normally I don't select any descriptor.

` service->writeCharacteristic(characteristic, value); `

but you write to the descriptor maybe this is the problem.

dsm, 2024-11-18:
You can look what is the purpose of the function [writeDescriptor ](https://doc.qt.io/qt-6/qlowenergyservice.html#writeDescriptor). We need to write and read data using selected characteristic, you connected characteristicRead to onCharacteristicChanged slot(assume reading working)

dsm, 2024-11-18:
I changed code in this commit f0560fb800921c3d126978abb8ff2717e2e52be1  every things should works but I think you should remove descriptor selection and start notify automatically or you should give us a button than do notify on selected characteristic.
```cpp
 -m_service->writeDescriptor(descr, data);
 +m_service->writeCharacteristic(characteristic, data);
```

alex-spataru, 2024-11-18:
Thanks so much for the PR! I’ll go ahead and remove the descriptor selection from the UI. Honestly, I have limited experience with BLE, so this has been a learning experience for me. While testing with other software, I noticed that descriptors are typically “hidden” from the user, so it makes sense for us to align with that approach.

[image]

dsm, 2024-11-18:
I was normally writing to the characteristic and reading it with python, but I thought it was like that in your code, I saw it when I looked at it. it seemed strange to choose a descriptor. In my free time (I'm currently employed and also studying for a master's degree.) I can make improvements and developments on the ble side(or serial side), I had qml experience before but I am not a UI guy :).

alex-spataru, 2024-11-18:
@dsm Hi! I just added a commit that removes the descriptors list from the UI. Now, the CCCD is automatically set to notify as soon as a characteristic is selected. Could you check if the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) works for you as well? If it does, we can close this issue.

> I was normally writing to the characteristic and reading it with Python, but I thought it was like that in your code. I saw it when I looked at it. It seemed strange to choose a descriptor.

Haha, BLE is definitely intimidating when you’re used to simpler protocols like serial, network, or USB. Still learning here! 🤣

> In my free time (I’m currently employed and also studying for a master’s degree.) I can make improvements and developments on the BLE side (or serial side).

Thank you so much! Any help with maintaining or adding new features to the project is always greatly appreciated. I’m currently self-employed and building a small company, so balancing everything can be a challenge. Collaborating with contributors like you makes a huge difference! Good luck on your masters!

If you’re looking for a fun non-UI problem to tackle in your free time, I’d recommend checking out the [CircularBuffer](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/CircularBuffer.h) and the threaded [FrameReader](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/FrameReader.h) implementations.

The **CircularBuffer** is used to temporarily store data for frame extraction and the console buffer, while the **FrameReader** runs on a separate thread and identifies frames in the incoming data stream. In my test cases, both seem to work fine, but I’d greatly appreciate additional external testing before rolling out a new release.

Another area that could use some attention is the [FileTransmission](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/FileTransmission.h) class, which ties closely to your comments in #225. It’s currently quite limited but supports straightforward CSV file transmission to a connected device. Any improvements or suggestions there would be incredibly helpful. Just keep in mind that I’d like to draft a new release this week, so minimizing the number of changes for stable users is a priority. This approach will also make it easier to keep the documentation and examples up to date.

Thanks again for your support and contributions!

alex-spataru, 2024-11-19:
Thank you very much for the feedback! Closing...

closed 2024-11-19 by alex-spataru