Serial-Studio/Serial-Studio #221: Multi plot no legend

yannislagnel, 2024-11-15:
## Describe the bug

Hello,
Tested on macOS 15 and windows 10
With version 2.0 or less when using "multiple plot" we have the legend for each data serie 
But  with versions >=3.0 I didn't find any legend and it is not possible to identify each data serie 

### Screenshots
Version 2.0
[image]
version 3.04
[image]

## Upload your JSON map file*

This is important in order to reproduce any UI-related issues.

[Untitled Project.json](https://github.com/user-attachments/files/17778564/Untitled.Project.json)

## Upload minimal Arduino program to replicate your issue

This is important so that I can replicate the environment and setup that causes the bug. 

An example program could be:
   Serial.print(">");
    Serial.print(tot_angle);
    Serial.print(",");
    Serial.print(u);
    Serial.print(",");
    Serial.print(error);
    Serial.print(",");
    Serial.print(pwmL);
    Serial.print(",");
    Serial.print(pwmR);
    Serial.print(",");
    Serial.print(desired_angle);
    Serial.print(",");
    Serial.print(kp);

    Serial.print("\n");
If you need me to test with a specific microcontroller, please consider [donating](https://www.paypal.com/donate?hosted_button_id=XN68J47QJKYDE) so that I can buy that MCU and help you!

alex-spataru, 2024-11-16:
Hi! Just fixed this issue, you can try it out in the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/continuous). You can also now enable/disable showing legends on multiple plot widgets using the "Show Legends" checkbox in the widget setup panel:

[image]

yannislag, 2024-11-17:
Hi Alex, 
Thank you for your reactivity.
I installed this version and It works very well.
Thank you again. 
Yannis

alex-spataru, 2024-11-18:
You're welcome! Closing this issue to keep things clean...

commit e44a1da0 (2024-11-16): Fix #221
closed 2024-11-18 by alex-spataru