Serial-Studio/Serial-Studio #224: parse and plot arbitrary text lines

lidavid, 2024-11-17:
Hey, 

My serial output is a log file where each line is a textual log message. I would like to know if there is a possibility with serial-studio to parse each line to get specific values from it an plot them. I tired with the Frame parser function but it seems like I am not able to achieve what I want.

For example if I have line like the following:
`log line 1 with value 5`
`log line 2 with value 10`

I want to be able  to parse each line and plot the values 5,10,....

tarman3, 2024-11-17:
In field **Separator**  input space symbol.
So you will get array ["log","line","1","with","value","-14"]

Your values have index 6.

lidavid, 2024-11-17:
It doesn't really work for me. doesn't it look for the start and end delimiter ?

tarman3, 2024-11-17:
What do you see in console? Show screenshot.

tarman3, 2024-11-17:
Easy ))

Frame Start Delimiter:  **value**
Frame End Delimiter: **log**

After parsing get array with two elements: ['"', "5"]
Your index = 2

[image]

tarman3, 2024-11-17:
Or set Frame Start Delimiter: **value** _with space character in the end_
Your index = 1

[image]

lidavid, 2024-11-17:
Thank you very much for your help ! What you suggest will defiantly work with the example I gave but I think I over simplified things.

My real log lines look like this:  `2000-01-01 01:26:21.195 <<<UNIT>>> - INFO - speed 10 rpm`
I would like to plot the speed value.

The first values of the date and time change for each log line and also there are other log lines in between that are of a different format ( they dont contain speed X rpm and also different log levels and units). What I know for sure is that each log starts in a new line '\n' and ends with a new line. 

What I would have liked to be possible is to get the full log line by the new line delimiters and parse it with javascript and output my specific values. but seems like it is not possible.

tarman3, 2024-11-17:
Frame Start Delimiter: **speed**_space_
Frame End Delimiter: _space_**rpm**

alex-spataru, 2024-11-17:
> I use UDP Socket to check this solution. But Network method have option Ignore data delimiters

Just to drop in the conversation, the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) allows users to ignore data delimiters, regardless of data source. Users will only need to check the "No Delimeters" checkbox. 

I plan to create a new release soon, I am just waiting for feedback to ensure that issue #220 has been resolved.

alex-spataru, 2024-11-17:
Another thing, maybe this issue can be solved by using [Regular Expressions](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Guide/Regular_expressions) in the frame parser, don't know if Qt's JavaScript engine supports it, but it might be worth giving it a try.

alex-spataru, 2024-11-17:
Just solved it! I’ve created a Python script that simulates your system by generating and transmitting data frames to a UDP socket. The script sends frames with key-value pairs such as speed, acceleration, temperature, and pressure. Below is the code:

### Python Simulator Script

```python
import socket
import random
import time

# UDP Configuration
UDP_IP = "localhost"
UDP_PORT = 9000

# Create a UDP socket
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

def generate_frame(key, value):
    """
    Generates a frame for a single key-value pair.
    Format: "timestamp <<<UNIT>>> - INFO - key value unit"
    """
    timestamp = time.strftime("%Y-%m-%d %H:%M:%S.%f")[:-3]
    return f"{timestamp} <<<UNIT>>> - INFO - {key} {value}"

def get_random_value(key):
    """
    Generates a random value based on the key.
    """
    if key == "speed":
        return f"{random.randint(0, 100)} rpm"
    elif key == "acceleration":
        return f"{random.uniform(0, 10):.2f} m/s^2"
    elif key == "temperature":
        return f"{random.uniform(20, 30):.2f} C"
    elif key == "pressure":
        return f"{random.randint(900, 1100)} hPa"
    return "0"

def send_frames():
    """
    Continuously generates and sends frames for each key to the specified UDP port.
    """
    keys = ["speed", "acceleration", "temperature", "pressure"]
    try:
        print(f"Sending frames to {UDP_IP}:{UDP_PORT}...")
        while True:
            for key in keys:
                value = get_random_value(key)
                frame = generate_frame(key, value)
                print(f"Sending: {frame}")
                sock.sendto(frame.encode(), (UDP_IP, UDP_PORT))
                time.sleep(0.1)  # Adjust the interval as needed
    except KeyboardInterrupt:
        print("\nStopped sending frames.")
    finally:
        sock.close()

if __name__ == "__main__":
    send_frames()
```

**Notes:**

- This script sends individual frames for each key-value pair, e.g., `2024-11-17 01:26:21.195 <<<UNIT>>> - INFO - speed 56 rpm`
- It uses a UDP socket to transmit frames to localhost on port 9000. You can adjust the IP address or port as needed.
- You can change the frequency of the transmissions by modifying the time.sleep(0.1) interval.

### JavaScript Frame Parser Code

Below is the JavaScript code to parse the frames and extract numerical values into an array, based on a user-defined mapping of keys to specific indices:

```javascript
// User-defined map to associate keys with specific indexes
// Update according to your sensor data and the indexes that you've set in the Project Editor 
const keyToIndexMap = {
    speed: 0,
    acceleration: 1,
    temperature: 2,
    pressure: 3
};

// Initialize the global array to hold parsed values (default to 0)
const parsedValues = new Array(Object.keys(keyToIndexMap).length).fill(0);

/**
 * Parses a frame, extracting values for keys defined in the keyToIndexMap,
 * and updates the parsedValues array at the specified index.
 *
 * @param[in]  frame      The latest received frame as a string.
 * @return     Array of numerical values corresponding to the keyToIndexMap.
 *
 * @example
 * Given frame: "2024-11-17 01:26:21.195 <<<UNIT>>> - INFO - speed 10 rpm"
 * Updates parsedValues: [10, 0, 0, 0]
 */
function parse(frame) {
    // Regular expression to match "key value" pairs
    const regex = /(\w+)\s+([\d.]+)/g;
    let match;

    // Iterate over all matches and populate the parsedValues array
    while ((match = regex.exec(frame)) !== null) {
        const key = match[1]; // Extract the key (e.g., "speed")
        const value = parseFloat(match[2]); // Extract the numeric value (e.g., 10)

        // Check if the key exists in the map and update the corresponding index
        if (keyToIndexMap.hasOwnProperty(key)) {
            const index = keyToIndexMap[key];
            parsedValues[index] = value;
        }
    }

    return parsedValues;
}
```

**Key Features:**
- **Key-to-Index Mapping:** Define how keys (e.g., speed, acceleration) map to array indices.
- **Real-Time Updates:** The `parsedValues` array is updated dynamically based on received frames.

### Screenshots

#### Console Output & UDP Socket Setup:

[image]

#### Real-Time Dashboard:

[image]

I’ve attached files for further experimentation:
- Project File: [Issue 224.json](https://github.com/user-attachments/files/17792215/Issue.224.json)
- Python Script (with .txt extension due to GitHub policies): [issue224.py.txt](https://github.com/user-attachments/files/17792229/issue224.py.txt)

You can use these files to replicate the setup or customize the project further. Let me know if you need additional guidance!

alex-spataru, 2024-11-17:
Another thing, in the next release (currently available in the continuous build), users will have the ability to configure delimiter settings directly in the Project Editor. This includes:
- **End Delimiter Only:** Ideal for scenarios where only an end delimiter (e.g., \n) is needed to process frames.
- **Start & End Delimiter:** Useful for projects that require both delimiters for more precise frame handling.

This enhancement greatly simplifies the project setup process and reduces configuration overhead, making it easier to adapt to various data input formats. I hope you find this new feature useful!

[image]