Serial-Studio/Serial-Studio #225: Process data from file

tarman3, 2024-11-18:
Idea to think about.

"Everything is a file" is an idea that Unix.
It might be worth adding the ability to process a file that is updated regularly.
It maybe some log file. Also, data from sensors can be obtained through a file.

Thanks for your work.

dsm, 2024-11-18:
@alex-spataru It should be good to send a file(maybe binary file) with specific chunk size to serial device. ex. teraterm has file transfer features that send file with specific protocol ex. (x/y/z)modem. It should be very useful to send a file common protocol or maybe if it's possible to send custom protocol. ex send a file with chunk size 120 byte wait for specific response or not and(or) send every 120 btyes CRC(8,16 or 32 configurable) or send overal file hash(sha,md5 or whatever) before file sent and(or) sending a file size before. file send dialog shows us a option what can we do a file and we can add a list to program file sending procedure ? I have no time to implement these but it is very use full something configurable and flexible.

alex-spataru, 2024-11-26:
@dsm Thanks for the feedback, is it possible for you to open another issue for file transmission so we can keep track of new feature requests? In this feature @tarman3 is basically proposing adding a new data source/driver for Serial Studio.

I think this is entirely possible as both the serial port and network port drivers use classes based on [QIODevice](https://doc.qt.io/qt-6/qiodevice.html), which is also the base class for [QFile](https://doc.qt.io/qt-6/qfile.html). I still need to investigate if we can open a file and get notified whenever the contents of the file change.

closed 2026-01-11 by tarman3