Serial-Studio/Serial-Studio #227: missing standard baudrate 921600 bps

mail87523, 2024-11-20:
missing standard baudrate 921600 bps for serial communication

mail87523, 2024-11-20:
The standard baud rates are the following: 4800, 9600, 19200, 38400, 57600, 115200, 230400, 256000, 460800, 576000,  921600

alex-spataru, 2024-11-25:
Hi, just fixed this issue in the latest commit. Thanks for the reminder!

**Quick tip:** if you need to input any custom baud rate, you can do so by typing directly over the baud rate list widget and pressing <enter>.

closed 2024-11-25 by alex-spataru

tarman3, 2024-11-25:
It's a pity that you can't manually enter the serial port address either.

alex-spataru, 2024-11-25:
> It's a pity that you can't manually enter the serial port address either.

The current setup makes adding a custom serial port path a bit cumbersome because we perform validations to filter out duplicate devices and devices that Qt flags as invalid (or null). I’ll look into adding support for this when I have the time.

- [IO::Drivers::Serial::validPorts()](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/Serial.cpp#L850)
- [IO::Drivers::Serial::open()](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Drivers/Serial.cpp#L163)

If you’re comfortable with C++ and Qt, you’re welcome to add support for this yourself and do the relevant testing for your test cases. One approach might be constructing a [QSerialPortInfo](https://doc.qt.io/qt-6/qserialportinfo.html#QSerialPortInfo) with a user-provided custom path. You’d also need to modify the QML interface code [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/qml/MainWindow/Panes/SetupPanes/Devices/Serial.qml#L74) to make the combobox editable and adding an `onAccepted` handler similar to the one used for the [baud rate combo](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/qml/MainWindow/Panes/SetupPanes/Devices/Serial.qml#L110). Feel free to submit a pull request if you get it working!

tarman3, 2024-11-25:
My limit is "Hello World" in Python ))

alex-spataru, 2024-11-25:
@tarman3 no problem, everyone of us starts with a Hello World program haha, I'll try to add support for that feature for the next release cycle.

For the moment I want to address the remaining reported bugs & focus on testing for v3.0.5. 

I'll begin working on this request and the general file I/O driver for 3.0.6.