Serial-Studio/Serial-Studio #228: MacOS build deployment target

dsm, 2024-11-20:
Hi, I don't see any deployment target for macos build and buildin macos 14 or 15 build shouldn't work on older macos. I normally build qt 6.8 app with minimum supported version macos 12 and Qt build can be easyly converted universal binary. if you want to support older ex for macos 10.14 for intel build you should be use qt 6.2.4. 

qt 6.8.0 supported version : https://doc.qt.io/qt-6/macos.html
qt 6.2.4 supported version :https://doc.qt.io/qt-6.2/macos.html

for:old macos
```bash
-DCMAKE_OSX_DEPLOYMENT_TARGET=10.14 \
-DCMAKE_OSX_ARCHITECTURES="x86_64" \
```

for universal latest binary.
```bash
-DCMAKE_OSX_DEPLOYMENT_TARGET=12.0 \
-DCMAKE_OSX_ARCHITECTURES="arm64;x86_64" \
```

dsm, 2024-11-20:
@alex-spataru generally universal binary almost double the size of binary in .app file, yes sticking with the latest qt version  sounds good.

alex-spataru, 2024-11-20:
Creating a universal binary in a single run becomes quite complex due to the [SIMDe](https://github.com/simd-everywhere/simde) dependency we use for real-time plotting data processing. While I'm still working on getting OpenMP linking to work correctly on macOS, the performance difference is already very noticeable on Windows and Linux. For more info, check the [CMakeLists.txt](https://github.com/Serial-Studio/Serial-Studio/blob/master/lib/simde/CMakeLists.txt) file that interfaces with SIMDe, it's still pretty much a work in progress though.

You can stress test Serial Studio using the [UDP Function Generator](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/UDP%20Function%20Generator) by testing with various frame intervals. It's important to note that when interacting with microcontrollers or remote devices (via BLE, network, or MQTT), achieving sub-millisecond frame intervals is quite difficult. That said, running the UDP Function Generator example locally is still a good way to identify potential bugs and bottlenecks in Serial Studio.

Anyways...For now, I've created separate builds for macOS—one for ARM64 and another for Intel. Both are built on macos-latest and target macOS 12. I'll test later if the newly generated Intel build works as intended.

dsm, 2024-11-20:
I think you can do something like that. I try to stress Serial Studio thanks.

```
# Architecture-specific compiler flags
# ARM64: Enable NEON
set(ARM64_FLAGS "-mfpu=neon -ftree-vectorize")
# x86_64: Enable AVX/SSE
set(X86_64_FLAGS "-msse4.2 -mavx -ftree-vectorize")

# Apply flags based on architecture
target_compile_options(my_program PRIVATE
    $<$<ARCHITECTURE:arm64>:${ARM64_FLAGS}>
    $<$<ARCHITECTURE:x86_64>:${X86_64_FLAGS}>
)
```

alex-spataru, 2024-11-26:
> I tested latest build without any issue. You can close the issue :)

Thanks! Closing issue...

closed 2024-11-26 by alex-spataru

alex-spataru, 2024-12-01:
@dsm Just managed to setup the CI workflow to provide universal builds on macOS. Tested with my devices and everything works well, let me know if you have any problems on your devices. Thanks!