Serial-Studio/Serial-Studio #229: 在十六进制发送模式下, 在末尾之前的任何一个地方添加/删除字符,光标都将被自动移动到末尾

Tinyu-Zhao, 2024-11-21:
### **对于中国用户 / For Chinese users, please delete this part if you speak English**

### **1. 描述问题**

> 请清晰简洁地描述您遇到的问题。  

[在十六进制发送模式下, 在末尾之前的任何一个地方添加/删除字符,光标都将被自动移动到末尾]

### **2. 截图（如果适用）**

> 如果适用，请添加截图以帮助说明您的问题。  

[

https://github.com/user-attachments/assets/ec244566-cc87-46cd-bb04-bc69b623f026

]

### **3. 复现方法**
切换到十六进制输入模式, 在末尾之前的任何一个地方添加/删除字符

alex-spataru, 2024-11-24:
I’ve updated the HEX validation code with the following improvements:

1. Ensured only valid HEX characters (0-9, A-F) are allowed in the input.
2. Added a check to highlight any incomplete or incorrectly formatted HEX sequences.
3. Enhanced the handling of cursor position to preserve it seamlessly when the text is updated, including special handling for edits made at the end of the input.

Please see the attached video for a demonstration. You can test these changes using the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous). Let me know how the new build performs on your end and if I should proceed with closing the issue. Thank you!

---

我已经对 HEX 验证代码进行了以下改进：
	1.	确保输入中只允许有效的 HEX 字符（0-9 和 A-F）。
	2.	增加了对不完整或格式错误的 HEX 序列进行高亮显示的检查。
	3.	改进了光标位置的处理，在更新文本时能够无缝保留光标位置，尤其针对在输入末尾进行的编辑做了特殊处理。

请查看附件中的视频以了解具体演示。您可以使用最新的[持续构建版本](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)测试这些更改。请告诉我新版本在您那边的表现，并告知是否可以关闭该问题。谢谢！

---

https://github.com/user-attachments/assets/8acbdea2-7611-494f-964f-36d4edc6ce84

Tinyu-Zhao, 2024-11-25:
谢谢你的答复, 这个问题得到改善, 但是我发现了另一个存在的问题

具体可以体现在视频的13秒处,当我在数字5后面输入7,再输入8,光标理应在数字8的后边,但是他却跑到了数字8的前面, 导致我再输入9变成了98

https://github.com/user-attachments/assets/4861df38-93a1-4de0-945f-ebc9b89acdc2

Tinyu-Zhao, 2024-11-25:
> 谢谢你的答复, 这个问题得到改善, 但是我发现了另一个存在的问题
> 
> 具体可以体现在视频的13秒处,当我在数字5后面输入7,再输入8,光标理应在数字8的后边,但是他却跑到了数字8的前面, 导致我再输入9变成了98
> 
>  Screen.Recordin1.2024-11-25.at.09.41.29.mov

我使用的是最新的[持续构建版本](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous)

alex-spataru, 2024-11-25:
You're welcome! Again, thanks for brining this issue to my attention. Just added another commit that should fix the remaining bug. As with the last response, the [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) should be available within 10-15 minutes of this message. Please let me know if I should close the issue!

---

不客气！再次感谢您让我注意到这个问题。我刚刚添加了一个新的提交，应该可以修复这个遗留的错误。与上次一样，[持续构建版本](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) 应该会在本消息发布后的 10-15 分钟内可用。请告诉我是否可以关闭该问题！

Tinyu-Zhao, 2024-11-26:
谢谢你,问题得到了解决👍

---

Thank you, the problem has been solved👍

closed 2024-11-26 by Tinyu-Zhao