Serial-Studio/Serial-Studio #233: Allow users to specify serial port location

alex-spataru, 2024-11-26:
As discussed in #227 and #73, this feature could be useful for adding support for virtual serial ports in GNU/Linux. Opening this issue to keep track of that request.

TODO's:
- Modify QML interface elements to allow user to type of serial port selector combo.
- Modify IO::Driver::Serial to support this feature.
- Check if this issue is possibly related with #225, don't know if virtual serial ports behave the same way as "normal" serial ports application-wise.
- Check if we should only enable this in UNIX-based systems (macOS, Linux) and disable it on Windows.

alex-spataru, 2024-11-27:
Hi @tarman3,

I’ve just committed an experimental implementation of this feature. A new [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) should be available for testing in the next 10–15 minutes.

Please give it a try and let me know how it works for you. I’ll continue refining the implementation based on your feedback and address any issues mentioned in this thread until it's ready for the next release cycle.

Looking forward to hearing your thoughts!

tarman3, 2024-11-27:
I naively thought that this would allow the use of a virtual serial port.
I was wrong.

I created a virtual port using the [socat](https://www.baeldung.com/linux/make-virtual-serial-port#3-create-serial-port-with-baud-rate).

```
socat -d -d pty,rawer,echo=0,link=/tmp/serial/ttyV0,b9600 pty,rawer,echo=0,link=/tmp/serial/ttyV1,b9600
```
The virtual serial ports are working. I can send messages between them in the terminal.
But in the Serial-Studio when I click the Connect button, nothing happens.
No messages, notifications or errors.

https://github.com/user-attachments/assets/01dfd106-480c-4df3-9da6-df17401e0653

alex-spataru, 2024-11-27:
I just added a commit that logs displays any serial port error instead of just closing the serial port automatically.

I think the issue is that Qt tries to _lock_ the virtual serial port for exclusive use (which is necessary for safe usage of real serial devices).

Maybe we can just add a flag to ignore resource acquisition & locking errors when connecting to custom serial ports.

alex-spataru, 2024-11-27:
I just added the option to ignore this error, and on my end I manage to stay connected to a port created using `socat`. I still have to test if data transmission is successful, I don't have access to a serial device right now. A new [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) should be available for testing in the next 10–15 minutes.

tarman3, 2024-11-27:
It is working like a magik )))

alex-spataru, 2024-11-27:
> It is working like a magic )))

Glad to hear that! I’ll keep this issue open for now to monitor and address any potential bugs or issues related to these modifications before creating a new release.

I also think it would be helpful to write a small example demonstrating how to use Serial Studio with `socat` on Linux/UNIX/macOS for future reference. My experience with `socat` is still quite limited—I only installed it today via `homebrew` on my development machine 😆.

tarman3, 2024-11-27:
I can make additions to the LTE example

alex-spataru, 2024-11-27:
That would be fantastic, thank you! When you have time, feel free to add a PR to the LTE example — no need to rush.

tarman3, 2024-11-28:
Please have a look
https://github.com/tarman3/Serial-Studio/tree/master/examples/LTE%20modem

alex-spataru, 2024-11-28:
> Please have a look
> https://github.com/tarman3/Serial-Studio/tree/master/examples/LTE%20modem
> Ready to create PR.

Hi, looks good to me! Feel free to proceed and create the PR whenever you’re ready.

closed 2025-04-21 by alex-spataru