Serial-Studio/Serial-Studio #235: After migrate from 3.0.4 to 3.0.5 Serial Studio does not start

tarman3, 2024-11-27:
After migrate from 3.0.4 to 3.0.5 Serial Studio does not start.
Showing with message. After click **Ok** nothing happened.

[image]

Checked with package from Archlinux repository and AppImage from github.
 
To solve the problem I did the following:
- Renamed folder `/home/user/Documents/Serial Studio`
- Run and closed SerialStudio.
- Rename folder to original name
- Run and closed SerialStudio

alex-spataru, 2024-11-27:
Hi @tarman3 

Does running Serial Studio on a terminal as:

`export QT_QPA_PLATFORM=xcb ./Serial-Studio-3.0.5-Linux-x86_64.AppImage `

Avoid the crash? I think this might be a dependency issue in which the wayland plugin is not distributed with the AppImage.

tarman3, 2024-11-27:
https://github.com/user-attachments/assets/2566373f-37e2-42b6-89fb-fd7b35571796

alex-spataru, 2024-11-27:
Just managed to replicate the issue.

To be sure that the program crashes in the same part, let’s try debugging it with `gdb`. Please ensure you keep the current configuration and don’t delete the project file. Here’s how you can proceed:

**1. Open a terminal and execute the following command:** `gdb ./Serial-Studio-3.0.5-Linux-x86_64.AppImage`

**2. Then, inside the gdb prompt, type:** `run`
  This will begin executing the program. It might run painfully slow, so please be patient and let it proceed.

**3. If the application crashes, type:** `bt`
  This will generate a backtrace, showing the sequence of function calls that led to the crash.

**4. Please share the terminal output with me.** 
  It should look something like this:

```bash
$ gdb ./Serial-Studio-3.0.5-Linux-x86_64.AppImage
(gdb) run
... (application runs and encounters a crash)
(gdb) bt
#0  <function_name> at <file>:<line>
#1  <function_name> at <file>:<line>
#2  ...
(gdb) quit
```

Thanks in advance for your help! 😊

---

In my case, the backtrace is similar to: 

```bash
0   QtWidgets                     	       0x1057cb5a0 QScopedPointer<QObjectData, QScopedPointerDeleter<QObjectData>>::get() const + 0 (qscopedpointer.h:112) [inlined]
1   QtWidgets                     	       0x1057cb5a0 decltype(fp.get()) qGetPtrHelper<QScopedPointer<QObjectData, QScopedPointerDeleter<QObjectData>> const>(QScopedPointer<QObjectData, QScopedPointerDeleter<QObjectData>> const&) + 0 (qtclasshelpermacros.h:103) [inlined]
2   QtWidgets                     	       0x1057cb5a0 QTextEdit::d_func() const + 0 (qtextedit.h:29) [inlined]
3   QtWidgets                     	       0x1057cb5a0 QTextEdit::document() const + 0 (qtextedit.cpp:791)
4   Serial-Studio                 	       0x1049ad938 JSON::FrameParser::isModified() const + 16
5   Serial-Studio                 	       0x1049b4228 JSON::ProjectModel::saveJsonFile() + 56
6   Serial-Studio                 	       0x1049b3cdc JSON::ProjectModel::openJsonFile(QString const&) + 3196
7   Serial-Studio                 	       0x1049b0424 JSON::ProjectModel::ProjectModel() + 1104
8   Serial-Studio                 	       0x1049b1fa4 JSON::ProjectModel::instance() + 80
9   Serial-Studio                 	       0x10497bf38 Misc::ModuleManager::initializeQmlInterface() + 152
10  Serial-Studio                 	       0x1049de420 main + 1152
11  dyld                          	       0x1997f0274 start + 2840
```

alex-spataru, 2024-11-27:
I found the root cause of the issue. I forgot to verify if the pointer to the `JSON::FrameParser` instance is initialized in `JSON::ProjectModel::saveJsonFile()`. The problem arises because the `JSON::FrameParser` is initialized only when the QML user interface is displayed, while the legacy frame parser conversion procedure is executed beforehand. This causes the program to enter an undefined state, leading to the crash.

I’ve implemented a fix, and a new [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) should be available for testing in the next 10–15 minutes. Let me know if it resolves the issue for you too.

alex-spataru, 2024-11-27:
Just created an [emergency release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.6) with this fix. Again, thank you very much for reporting this issue!

tarman3, 2024-11-27:
I apologize for the late reply. Was far from the PC.
Checked with 3.0.6. **Problem solved**

closed 2024-11-27 by tarman3