Serial-Studio/Serial-Studio #236: Project Editor - Save as

tarman3, 2024-11-27:
Just a thought to think about.

Function <ins>Save as</ins> instead of or in addition would be useful.

[image]

alex-spataru, 2024-11-27:
Thanks for the observation!

Here are two possible approaches we could take to address this:

### 1. Slightly modify the current implementation

Keep the current implementation, but adapt the button’s icon and text to reflect its actual behavior:
- Display “Save As” when the project editor is working on a new, unsaved project.
- Display “Save” when the project editor is working on a project that has already been saved to disk. In this case, clicking the save button would save directly without displaying a “Save As” dialog.

### 2. Add Separate “Save” and “Save As” Buttons

Introduce distinct buttons for _Save_ and _Save As_. While this might feel more intuitive, it could create challenges for smaller screens in the future (e.g consider that we might also add more widget types for groups some day, and that different languages will produce different UI size requirements).

With the UI redesign I began in late June, my goal was to create an interface vaguely resembling the “Ribbon” UI found in some CAD software or Microsoft Office. However, adding small save buttons to the window caption would require using a [frameless window](https://doc.qt.io/qt-6/qt.html#WindowType-enum) with custom decorations. This approach will eventually lead to significant complications when targeting multiple operating systems, as I learned the hard way with Serial Studio 2.0.0—it quickly became messy and unmanageable.

What are your thoughts on these options?

tarman3, 2024-11-27:
I would like to have a separate button <ins>Save As</ins>, so that you can create a copy of the current project.

Now, to do this, you either need to create a new project from scratch or create a copy of the file through the file manager.

alex-spataru, 2024-11-27:
Just added a ‘Save As’ button, and the latest continuous build should be available shortly. We’ll let this issue open for a while, and if there are no objections, we’ll keep this change.

[image]

alex-spataru, 2025-01-17:
Hi @tarman3, hope you're doing well and wishing you a Happy New Year! Since there have been no objections to the change, I'll go ahead and close this issue...thanks!

closed 2025-01-17 by alex-spataru