Serial-Studio/Serial-Studio #237: UI Organization of connection methods

tarman3, 2024-11-28:
Just thoughts to think about.

Even when I first launched the program, I was a little confused by the presence of a separate MQTT button. At the same time, in the right area there is a menu for selecting all other connection methods. This may confuse a person who is launching the program for the first time.

Is this implemented for technical reasons and the complexity of implementation?
Otherwise, perhaps it is worth considering grouping all connection options in one menu.

alex-spataru, 2024-11-28:
The MQTT button is separate because Serial Studio isn’t just a subscriber—it can also act as a _publisher_. This allows it to forward data received through a serial connection (or TCP/UDP, Bluetooth, etc) to a MQTT broker, making it accessible to any other subscriber, including other Serial Studio instances or third-party software.

For example, I used this feature during CanSat competitions. Rockets and satellites transmitted telemetry via radio signals to a ground station device, which decoded the signals and sent the data to Serial Studio via a serial port. Serial Studio displayed a real-time dashboard while simultaneously publishing it to an MQTT broker. This allowed other software and users to connect as subscribers, including the competition organizers, who used proprietary software to log and cross-check the data remotely for validation.

Grouping MQTT with the main connection menu might confuse users who only want to forward data. However, improving the [MQTT wiki page](https://github.com/Serial-Studio/Serial-Studio/wiki/MQTT-Integration) with tutorials, videos or examples could make the publisher feature clearer, especially for first-time users.

tarman3, 2024-11-28:
> The MQTT button is separate because Serial Studio isn’t just a subscriber—it can also act as a _publisher_.

This is a key point that I didn't realize right away.
Now it's clear.

Thank You.

closed 2024-11-28 by tarman3