Serial-Studio/Serial-Studio #241: MacOS 15.1 Requires Sudo Mode to Receive data

zhenyulincs, 2024-12-03:
**Bug Report Template**

---

### **1. Describe the bug**

On macOS 15.1 only works properly when run in sudo mode. Without sudo privileges, the software stops receiving data from an ESP32 device after approximately 1 second. The ESP32 device also stop sending data.

Command:
```bash
sudo /Applications/Serial\ Studio.app/Contents/MacOS/Serial-Studio
```
### **2. Screenshots (if applicable)**

Without Sudo
[image]

With Sudo 
[image]

### **3. Upload your JSON map file**

Using "No Parsing" also cause the issue

### **4. Provide a minimal Arduino program to replicate the issue**

[ESP32 esp-idf]  

Here’s an example of a minimal program:  

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"
#include "driver/gpio.h"

#define LED GPIO_NUM_2                // GPIO pin for the LED
#define UART_PORT_NUM UART_NUM_0      // UART port (using UART0 for default serial communication)
#define UART_TX_PIN 1                 // TX pin for UART
#define UART_RX_PIN 3                 // RX pin for UART
#define BUF_SIZE 1024                 // UART buffer size

// Function to toggle the LED
void LED_toggle() {
    static uint8_t led_status = 0;    // LED state (0 = OFF, 1 = ON)
    led_status = !led_status;        // Toggle the state
    gpio_set_level(LED, led_status); // Set the LED level
}

// Initialize the LED
void LED_init() {
    gpio_reset_pin(LED);                 // Reset the GPIO pin
    gpio_set_direction(LED, GPIO_MODE_OUTPUT); // Set as output
    gpio_set_level(LED, 0);              // Start with LED off
}

// Initialize UART
void uart_init() {
    const uart_config_t uart_config = {
        .baud_rate = 115200,                  // Baud rate
        .data_bits = UART_DATA_8_BITS,        // 8 data bits
        .parity = UART_PARITY_DISABLE,        // No parity
        .stop_bits = UART_STOP_BITS_1,        // 1 stop bit
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE // No flow control
    };

    // Install UART driver with RX buffer only (no TX buffer needed)
    uart_driver_install(UART_PORT_NUM, BUF_SIZE * 2, 0, 0, NULL, 0);
    uart_param_config(UART_PORT_NUM, &uart_config); // Configure UART
    uart_set_pin(UART_PORT_NUM, UART_TX_PIN, UART_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}

// Main application
void app_main(void) {
    LED_init();  // Initialize LED
    uart_init(); // Initialize UART

    uint8_t data[BUF_SIZE]; // Buffer for UART data

    while (1) {
        // Read data from UART
        int len = uart_read_bytes(UART_PORT_NUM, data, BUF_SIZE, 20 / portTICK_PERIOD_MS);
        if (len > 0) {
            // Echo the received data back via UART
            uart_write_bytes(UART_PORT_NUM, (const char *)data, len);

            // Toggle LED state on receiving data
            LED_toggle();
            printf("%.*s\n", len, data);
        } 
    }
}

```

alex-spataru, 2024-12-03:
Hi!

I believe this issue is less about privileges and more about the speed at which data is being received, which can overwhelm Serial Studio. When running in `sudo` mode, Serial Studio tends to perform better because the operating system prioritizes the frame extraction thread/process as requested by Serial Studio [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Manager.cpp#L90). It also allows the GUI thread to run with higher priority [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/main.cpp#L102), leading to better handling of the data flow and a more responsive user interface under heavy loads (especially on Windows and Linux).

To help with diagnosing the issue, could you try adding a small delay (e.g., 1 ms, or any value greater than 0.1 ms) between data frames in your ESP32 program? Does that improve stability?

Additionally, could you let me know if you're using an Intel or arm64 processor? Most of my testing on macOS is done on an M2 chip, which works very well. However, this could make me unaware of optimizations that may be needed for older macOS systems.

zhenyulincs, 2024-12-03:
> Hi!
> 
> I believe this issue is less about privileges and more about the speed at which data is being received, which can overwhelm Serial Studio. When running in `sudo` mode, Serial Studio tends to perform better because the operating system prioritizes the frame extraction thread/process as requested by Serial Studio [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/IO/Manager.cpp#L90). It also allows the GUI thread to run with higher priority [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/src/main.cpp#L102), leading to better handling of the data flow and a more responsive user interface under heavy loads (especially on Windows and Linux).
> 
> To help with diagnosing the issue, could you try adding a small delay (e.g., 1 ms, or any value greater than 0.1 ms) between data frames in your ESP32 program? Does that improve stability?
> 
> Additionally, could you let me know if you're using an Intel or arm64 processor? Most of my testing on macOS is done on an M2 chip, which works very well. However, this could make me unaware of optimizations that may be needed for other macOS systems.

Thank you for your reply.
By adding 100ms delay at ESP32, the issue still persists.

Here is my system info
CPU: Apple M3 Chip
OS: MacOS 15.1 (24B83)

Note: The solution involving sudo permissions originates from the STM32 forum. STM32CubeIDE version 16.0 cannot flash code to the board unless it's in sudo mode. Additionally, this issue occurs only on MacOS 15.
Link: https://community.st.com/t5/stm32cubeide-mcus/stlink-server-with-macos-15/td-p/705089

alex-spataru, 2024-12-03:
Could you let me know the exact version of the ESP32 microcontroller you’re using? Additionally, have you tried enabling the [DTR](https://en.wikipedia.org/wiki/Data_Terminal_Ready) checkbox or adjusting the flow control options? If you make any changes in Serial Studio, please ensure the same adjustments are reflected in your ESP32 program to avoid mismatches.

I primarily use Chinese clone boards, as they’re the most accessible locally. If this issue turns out to be hardware-specific, I may need to order a genuine ESP32 from Amazon to investigate further.

For reference, while testing a budget NodeMCU ESP32 v1.1 clone from Steren, I noticed that Serial Studio stops receiving data when the DTR checkbox is **not** checked. Enabling the DTR checkbox resolves the issue for me:

https://github.com/user-attachments/assets/18ae9867-0473-44e7-8a3e-833fd345bd99

Let me know if adjusting the settings or identifying hardware differences makes a difference!

zhenyulincs, 2024-12-03:
> Could you let me know the exact version of the ESP32 microcontroller you’re using? Additionally, have you tried enabling the [DTR](https://en.wikipedia.org/wiki/Data_Terminal_Ready) checkbox or adjusting the flow control options? If you make any changes in Serial Studio, please ensure the same adjustments are reflected in your ESP32 program to avoid mismatches.
> 
> I primarily use Chinese clone boards, as they’re the most accessible locally. If this issue turns out to be hardware-specific, I may need to order a genuine ESP32 from Amazon to investigate further.
> 
> For reference, while testing a budget NodeMCU ESP32 v1.1 clone from Steren, I noticed that Serial Studio stops receiving data when the DTR checkbox is **not** checked. Enabling the DTR checkbox resolves the issue for me:
> 
>  Screen.Recording.2024-12-03.at.2.18.03.a.m.mov 
> Let me know if that fixes your issue!

Hi,

After check the DTR checkbox, the issue is fixed!

For people who has the problem in the future, I didn't change my esp32 code, only enable the DTR in the Serial Studio.

Here is my ESP32 board: [ESP-WROOM-32](https://www.amazon.com/ESP-WROOM-32-Development-Microcontroller-Integrated-Compatible/dp/B08D5ZD528/ref=sr_1_1?crid=3IDJH54HJME46&dib=eyJ2IjoiMSJ9.C2zjybnvjkLUB8ybNqfUPfgJ8K4W3E0I-8GYneZAZtYTGG2tMx8ZhYZ7kWXPXIX9YqE-D6lRmsU2uw1UFblkL7tkn63jHZoJ55KQ5I6CeidRaXvMHcKh-dhY-n9hapDVlhC0lwxsucvNgRWFzsONrXxc38giP45ByxUAkGUHHTvgwJXj-d4uXH_IM9bDCv3-A3ajSwGIbYoYvhSxvh6B9WesfF-e9zc2zn9yaMTrec0.-7fkrQsjOyMx2fMZl6e1F00oqLeRyc3IM9ZvAF8Sv3o&dib_tag=se&keywords=esp32+wroo%2C&qid=1733214835&sprefix=esp32+wroo%2C%2Caps%2C423&sr=8-1)

closed 2024-12-03 by zhenyulincs