Serial-Studio/Serial-Studio #255: Created Homebrew cask for macOS -- would you like to review before submission?

davay, 2024-12-29:
First, apologies if this is not the right place to discuss this, especially since it didn't fit any of the templates.

So I've just created a Homebrew cask to install Serial-Studio on MacOS that:
- Works on both Intel and Apple Silicon Macs
- Should stay in sync with new GitHub releases, provided the naming scheme doesn't change

Commit: https://github.com/Homebrew/homebrew-cask/commit/109a87940634702692e3e21601ddd40d601d893f 

I think it's ready for submission, but I thought I should at least notify you as the creator first.

To test this I've done: 
- `HOMEBREW_NO_INSTALL_FROM_API=1 brew install --build-from-source serial-studio`: test installation (on apple silicon only though)
- `brew audit --new serial-studio`: per requirements
- `brew style --fix serial-studio.rb`: best practices
- `brew livecheck serial-studio --debug`: test auto-update
- Checked existing casks/PRs (just to see if somebody's already/currently making one)

Technically Homebrew is supported on Linux too, but I feel people prefer other native repos anyway, let me know if you'd like me to add that too or if this is good to go.

Thanks 🙂

alex-spataru, 2025-01-02:
Hi @davay,

First of all, thank you so much for the work you've done! I use Homebrew a lot, but I never found the time to register my projects with it, so I really appreciate you stepping in to do that. The current code looks great—thanks again!

Regarding Linux, I’ve never used Homebrew on it either. Like you mentioned, I tend to stick with distro-specific repositories or just install the .deb/.rpm packages directly. But if you think there’s value in having it available through Homebrew on Linux, feel free to add it.

Thanks again,
Alex

davay, 2025-01-10:
@alex-spataru The [PR](https://github.com/Homebrew/homebrew-cask/pull/197310) is checked in! 

Here's the cask: https://formulae.brew.sh/cask/serial-studio

No problem, it wasn't much, otoh I think the amount of support you provide is amazing, not really a Chinese speaker but this is the first project I've seen where there's support for another language even though you don't speak it yourself either

Best of luck! It's a great software 🙂

closed 2025-01-10 by davay